NVIDIA/jetson-gpio #1: How to use this for servos / sensors?

msurguy, 2019-04-02:
Hi! Thanks for this library. I'm just getting started with Jetson Nano and was wondering how I can use this (or another) library to interface with servos and I2C sensors. 

Any pointers would be appreciated. Thanks!

atan1021, 2019-04-03:
The Jetson.GPIO library does not support direct use of the I2C pins on the Jetson boards. 

However, the library is designed so that you can directly copy/paste working code on a Raspberry Pi that uses the RPi.GPIO library onto the Jetson and the code will still work. Most tutorials found online made for the Raspberry Pi should also work on the Jetson. Note that there is no hardware PWM for the Jetson. Software PWM (not integrated in the library) is possible, but given that the Jetson runs on Linux, real-time applications are not ideal.

For I2C capabilities, there is integration with Adafruit hardware using their CircuitPython library - Adafruit_Blinka.
Here is an example of an i2c servo driver and how to interface with it, which could be useful to you:
https://learn.adafruit.com/16-channel-pwm-servo-driver/python-circuitpython

Hope this helps!

tsaoyu, 2019-05-02:
The `Jetson.GPIO` alias to `RPi.GPIO` is pretty neat. 

Is there any hardware or hardware-timed PWM GPIO implementation similar to [pigpio](http://abyz.me.uk/rpi/pigpio/)?

swarren, 2019-05-02:
Sorry, there's no PWM support.

Since this is a bug tracker, and this entry doesn't report a bug, I'm going to close it. General questions can be made/answered in our web forum: https://devtalk.nvidia.com/default/board/139/embedded-systems/1. Thanks.

closed 2019-05-02 by swarren