NVIDIA/jetson-gpio #2: Issue with virtualenv

VincentDuf, 2019-05-03:
Hi,
I followed the process to install the library but I'm also using virtualenv to separate my workspaces.
So I did pip install in my virtualenv and when I want to import Jetson.GPIO it tells me I don't have the permissions:
```
import Jetson.GPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/nvidia/.virtualenvs/cv/lib/python3.5/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/home/nvidia/.virtualenvs/cv/lib/python3.5/site-packages/Jetson/GPIO/gpio.py", line 33, in <module>
    raise RuntimeError("The current user does not have permissions set to "
RuntimeError: The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this
```
Even after I had gpio as a group and included nvidia as a new member and copy and reload the rules..

The problem then is if I want to use python as root , I then have the old environment for python where Jetson.GPIO is not installed of course...

Is there a way to solve this ?

Thank in advance

swarren, 2019-05-03:
I don't believe this issue has anything to do with virtual env per se; it's a simple case of not having permissions to access the relevant kernel files to control the GPIOs.

After adding your user to the GPIO group, did you log out and log in to pick up that change?

What do the following commands output:
ls -lFa /sys/class/gpio
id -a

swarren, 2019-05-15:
yishuifengxing, please see https://github.com/NVIDIA/jetson-gpio/blob/master/README.md, which describes how to set up the permissions required for this code to work.

yishuifengxing, 2019-05-15:
我已经看过那篇文章了，也按照他的方法设置了。但是还是不能成功。不知道是不是由于我的python环境是按照安装在虚拟环境中导致的呢

yishuifengxing, 2019-05-15:
> yishuifengxing，请参阅https://github.com/NVIDIA/jetson-gpio/blob/master/README.md，其中介绍了如何设置此代码工作所需的权限。

我已经看过那篇文章了，也按照他的方法设置了。但是还是不能成功。不知道是不是由于我的蟒蛇环境是按照安装在虚拟环境中导致的呢

swarren, 2019-05-15:
(Google translates that previous comment as:)
> I have already read the article and set it up according to his method. But still can't succeed.
> I don't know if it is because my python environment is caused by installing in a virtual
> environment.

yishuifengxing, I do not believe this issue is related to virtualenv. Perhaps you can try to reproduce the problem without using virtualenv to confirm this. Please use the NVIDIA forums for support; https://devtalk.nvidia.com/default/board/139/embedded-systems/1 Thanks.

yishuifengxing, 2019-05-15:
> （谷歌将之前的评论翻译为:)
> 
> > 我已经阅读了这篇文章并根据他的方法进行了设置。但仍然无法成功。
> > 我不知道是不是因为我的python环境是由在虚拟
> > 环境中安装引起的。
> 
> yishuifengxing，我不相信这个问题与virtualenv有关。也许您可以尝试重现问题，而无需使用virtualenv来确认这一点。请使用NVIDIA论坛获取支持; https://devtalk.nvidia.com/default/board/139/embedded-systems/1谢谢。

我重启了Jetson TX2之后可以了呢，可能我之前设置好之后，没有生效。重新启动后，Jetson.GPIO就可以使用了。
谢谢您的回复。

swarren, 2019-05-15:
> After I restarted the Jetson TX2, I can do it. Maybe it didn't take effect after I set it up.
> After restarting, Jetson.GPIO will be available.

Thanks.

I've also confirmed that Jetson.GPIO works correctly inside a Python2 virtualenv on Jetson AGX Xavier.

@VincentDuf does a reboot fix the problem for you? If so, we can close this bug.

VincentDuf, 2019-05-16:
Hi, sorry for the delay I was out of office.

Here are the result of the 2 commands you asked:

```
nvidia@tegra-ubuntu:~$ ls -lFa /sys/class/gpio/
total 0
drwxr-xr-x  2 root root    0 May 16 09:28 ./
drwxr-xr-x 73 root root    0 Feb 11  2016 ../
--w--w----  1 root gpio 4096 Feb 11  2016 export
lrwxrwxrwx  1 root root    0 Feb 11  2016 gpiochip216 -> ../../devices/bpmp_i2c/i2c-4/4-003c/max77620-gpio/gpio/gpiochip216/
lrwxrwxrwx  1 root root    0 Feb 11  2016 gpiochip224 -> ../../devices/3160000.i2c/i2c-0/0-0077/gpio/gpiochip224/
lrwxrwxrwx  1 root root    0 Feb 11  2016 gpiochip240 -> ../../devices/3160000.i2c/i2c-0/0-0074/gpio/gpiochip240/
lrwxrwxrwx  1 root root    0 Feb 11  2016 gpiochip256 -> ../../devices/c2f0000.gpio/gpio/gpiochip256/
lrwxrwxrwx  1 root root    0 Feb 11  2016 gpiochip320 -> ../../devices/2200000.gpio/gpio/gpiochip320/
--w--w----  1 root gpio 4096 Feb 11  2016 unexport
nvidia@tegra-ubuntu:~$ id -a
uid=1001(nvidia) gid=1001(nvidia) groups=1001(nvidia),4(adm),20(dialout),24(cdrom),25(floppy),27(sudo),29(audio),30(dip),44(video),126(i2c),999(gpio)
nvidia@tegra-ubuntu:~$ 
```
The reboot didn't fix the issue

Thanks

swarren, 2019-05-16:
@VincentDuf can you please confirm which type of Jetson (TX1, TX2, AGX Xavier, or Nano) you're using, and which version of L4T you have installed (the first line in `/etc/nv_tegra_release` will tell you the L4T release).

Your error is very puzzling, since the file permissions appear to be correct. Can you please enter the following commands into the Python interactive interpreter:

```
import os
_SYSFS_ROOT = "/sys/class/gpio"
print(os.access(_SYSFS_ROOT + '/export', os.W_OK))
print(os.access(_SYSFS_ROOT + '/unexport', os.W_OK))
```

Both of those last two commands should print `True`, but given the error you're getting, it seems they are returning a different value for some reason. I'd like to confirm this; then we can think about how to debug the issue.

yishuifengxing, 2019-05-17:
> 嗨，抱歉我不在办公室。
> 
> 以下是您询问的2个命令的结果：
> 
> ```
> nvidia@tegra-ubuntu:~$ ls -lFa /sys/class/gpio/
> total 0
> drwxr-xr-x  2 root root    0 May 16 09:28 ./
> drwxr-xr-x 73 root root    0 Feb 11  2016 ../
> --w--w----  1 root gpio 4096 Feb 11  2016 export
> lrwxrwxrwx  1 root root    0 Feb 11  2016 gpiochip216 -> ../../devices/bpmp_i2c/i2c-4/4-003c/max77620-gpio/gpio/gpiochip216/
> lrwxrwxrwx  1 root root    0 Feb 11  2016 gpiochip224 -> ../../devices/3160000.i2c/i2c-0/0-0077/gpio/gpiochip224/
> lrwxrwxrwx  1 root root    0 Feb 11  2016 gpiochip240 -> ../../devices/3160000.i2c/i2c-0/0-0074/gpio/gpiochip240/
> lrwxrwxrwx  1 root root    0 Feb 11  2016 gpiochip256 -> ../../devices/c2f0000.gpio/gpio/gpiochip256/
> lrwxrwxrwx  1 root root    0 Feb 11  2016 gpiochip320 -> ../../devices/2200000.gpio/gpio/gpiochip320/
> --w--w----  1 root gpio 4096 Feb 11  2016 unexport
> nvidia@tegra-ubuntu:~$ id -a
> uid=1001(nvidia) gid=1001(nvidia) groups=1001(nvidia),4(adm),20(dialout),24(cdrom),25(floppy),27(sudo),29(audio),30(dip),44(video),126(i2c),999(gpio)
> nvidia@tegra-ubuntu:~$ 
> ```
> 
> 重启没有解决问题
> 
> 谢谢
通过pip命令安装后，在/opt/nvidia的路径下面是没有jetson-gpio文件夹的，需要在GitHub下载，并解压到/opt/nvidia
路径下面，然后再根据https://github.com/NVIDIA/jetson-gpio/blob/master/README.md   这里的文档设置权限，设置完之后重新启动就可以了呢。我这边是这样做的。

VincentDuf, 2019-05-17:
Hi,
I'm working on Jetson TX2 (bought 1 year ago in May 2018). 
Here is the version installed:
`R28 (release), REVISION: 2.1, GCID: 11272647, BOARD: t186ref, EABI: aarch64, DATE: Thu May 17 07:29:06 UTC 2018`

I executed your command inside and outside the virtualenv using the same version of python and it always returned "True"...

yishuifengxing, 2019-05-17:
您需要安装这两个东西
sudo apt-get install virtualenvwrapper
sudo apt-get install virtualenv virtualenvwrapper

安装好后您在新建虚拟环境的时候会报错，就像下面一样：
问题：-bash: mkvirtualenv: command not found

然后您需要进行下面几个操作：
1.创建目录用来存放虚拟环境
    mkdir $HOME/.virtualenvs
2.在~/.bashrc中添加行：
    export WORKON_HOME=$HOME/.virtualenvs
3.运行:
    source ~/.bashrc

swarren, 2019-05-17:
@VincentDuf are you able to upgrade to L4T release 32.1? The Jetson.GPIO library was designed on/for that L4T release, although admittedly it should work just fine on older releases too, and the issue you're seeing doesn't immediately seem related to the L4T version.

> I executed your command inside and outside the virtualenv using the same version of python and it
> always returned "True"...

That is very odd. Are you able to edit gpio.py and split apart the check at line 33 so you can determine what is failing and why? The check at gpio.py:33 is exactly what I had you run manually, which passed, so I'm confused why it fails when run as part of gpio.py.

VincentDuf, 2019-05-20:
Nope I can't cause I'm using a camera from e-con system only compatible with Jetpack3.2/3.3 so it's not possible for me.
I don't get your question cause line 33 from gpio.py I have "_pyset = set" and no tests...

swarren, 2019-05-20:
Your copy of gpio.py isn't the same as the copy at the following link?
https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio.py

VincentDuf, 2019-05-22:
Oups sorry you're right, it's from another repository actually ! I tried yours and it didn't work for some reason (the permission issue) but then a colleague came and used another one.
I'll ask you back asa I asked him 
Thank you

swarren, 2019-06-17:
I assume this is resolved, so closing.

closed 2019-06-17 by swarren

VincentDuf, 2019-06-18:
Hi Warren, 
sorry for the delay, I found another way to use the gpio by importing a script in my main.
The matter is resolved.

Thanks for support