NVIDIA/jetson-gpio #3: cannot find devices

Q-Z-one, 2019-05-05:
I do everything as the readme file say, but the terminal returns the following err
`No such file or directory: '/sys/devices/3160000.i2c/i2c-0/0-0074/gpio`

I reboot jetson but it seems that doesn't work, btw, I have run the sample successfully on the same jetson just yesterday.

swarren, 2019-05-06:
Please obtain support from the web forum linked at https://developer.nvidia.com/embedded/community/support-resources.

> btw, I have run the sample successfully on the same jetson just yesterday.

It sounds like there's no issue with Jetson.GPIO itself, then.

swarren, 2019-05-15:
@Q-Z-one did you resolve this issue?

swarren, 2019-05-20:
I assume this is not an issue in the Jetson.GPIO since it worked once. Hence, closing. If this issue still persists, and it's an issue with the library itself, feel free to re-open this bug.

closed 2019-05-20 by swarren