NVIDIA/jetson-gpio #4: Unstable performance of GPIO port + motion sensor

bittnt, 2019-05-11:
I use the same motion sensor. When plug in GPIO port of Raspberry Pi 3 B+, it works fine. When I use Jetson Nano and jetson-gpio library, it performs differently, it seems to be very unstable and produces high and low very frequently. 

What could be problem between the jetson-gpio and the one used in Raspberry pi given that I use the (almost) same code and same sensor. Is there any bug that would affect this in the jetson-gpio?

swarren, 2019-05-15:
What motion sensor are you interfacing with? Can you specify exactly what is going wrong in your case?

The low-level mechanism that Jetson.GPIO uses (sysfs GPIO files) may yield lower performance than the mechanism used by RPi.GPIO on an RPi; (optional?) direct HW register access. It is possible this timing change will negatively interact with some devices. There is currently no solution available within Jetson.GPIO for such issues if they do arise, but you could write a Linux kernel driver instead.

kylezhengai, 2019-05-16:
For example: https://www.adafruit.com/product/189?gclid=EAIaIQobChMI4fyexfeg4gIVDv5kCh10hAZOEAQYASABEgKR5fD_BwE

We tested the same sensor on Raspberry Pio using RPi.GPIO, we expect to get 'true' if there is motion detected; and 'false' if there is no motion detected. 

We tried the same code and the sample code provided by jetson-gpio with this sensor plugged in, the motion sensor gives not very consistent results compared to the same experiment with the same device on Raspberry Pi. We did back and forth experiment, this experiment repeats several times.

We think there is a problem with the Jetson.GPIO use or this is due to the reason you mention. Could you provide a good detailed fix. I am not sure how to write a Linux Kernel driver.

swarren, 2019-05-16:
That particular device has a simple single signal output. Therefore, it won't be affected by the performance of the GPIO library, since there's no complicated protocol to bit-bang or parse. The most the performance will affect is the maximum polling rate.

It sounds like you have some kind of electrical issue that's unrelated to Jetson.GPIO. You could confirm this by accessing the GPIO sysfs files or GPIO HW registers directly and seeing if the issue same issue still occurs. The best place to get support for Jetson itself as opposed to the Jetson.GPIO library specifically is the Jetson forum; https://devtalk.nvidia.com/default/board/139/embedded-systems/1.

kylezhengai, 2019-05-21:
Thanks. You can close this issue now. It turns out to be working now. I think the motion sensor (with 3.3 V power supply, not from adafruit) does not work, but the motion sensor (with 5 V power supply, from adafruit) works. Both sensors work fine on Raspberry Pi. I guess this is because the threshold setting in GPIO or the HW.

swarren, 2019-05-22:
OK, thanks.

closed 2019-05-22 by swarren