NVIDIA/jetson-gpio #5: pull_up_down value is unused

jhurliman, 2019-05-28:
In https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio.py#L218 the `pull_up_down` function parameter appears unused.

swarren, 2019-05-28:
It is indeed; there's no way to control pull-up/-down from user-space on Jetson.

swarren, 2019-06-17:
Since we have no way to resolve this, closing.

closed 2019-06-17 by swarren

dlarue, 2019-06-23:
Could you please add a comment as to how to control pull-up/pull-down GPIO resistors OUTSIDE of user-space?

joshberryhooves, 2019-06-26:
We have come into this issue with our base-board design for the Jetson Nano, with the same symptoms as described [here](https://devtalk.nvidia.com/default/topic/1055269/jetson-nano/gpio-button-seems-to-latch-maybe-wrong-resistor-value-but-unsure-of-cause/post/5349418/#5349418) Are there any plans to resolve this issue in the future (near or far), or do we need to work out a hardware work around?

jonhunter, 2019-06-27:
I have added some more commentary to the forum topic [here](https://devtalk.nvidia.com/default/topic/1055269/jetson-nano/gpio-button-seems-to-latch-maybe-wrong-resistor-value-but-unsure-of-cause/post/5349418/). Essentially it is possible to reconfigure the pull-up/down configuration, but this needs to be done by generating a new pin configuration for the device that needs to be flashed. We are working on putting some documentation together to assist with this.

swarren, 2019-07-01:
There are currently no plans to fix this issue in Jetson.GPIO itself.

dlarue, 2019-07-01:
Not surprising but hopefully it wakes up the hardware team to work at generalizing GPIO if they expect it to be more useful to the maker community.  The maker community expands and invents because they use things in ways few or no others thought of.

In the mean time, with the map of which pins are pulled up and down by what resistors is helpful. If there are ways to float or swap the polarity outside of GPIO library please point us to the HowTo on that since things like this are generally setup things and could be done outside in a shell before the applications run.

maxvfischer, 2020-11-21:
> It is indeed; there's no way to control pull-up/-down from user-space on Jetson.

Why are you even allowing the user to input the parameter `pull_up_down`, as well as raising errors when the wrong value is inputted, when you don't support the functionality? Am I missing something here? Cause this is super confusing for users of your package. I just spend 5 hours on this until i eventually checked the source code and found out that it wasn't used.

swarren, 2020-11-23:
@maxvfischer the Jetson.GPIO is an exact mirror of RPi.GPIO, and so takes all the parameters that it takes so that software originally written for RPi.GPIO can run on Jetson.GPIO without modification. Sorry for the confusion.

maxvfischer, 2020-11-23:
@swarren I understand, but the problem in this case is that Jetson.GPIO can't be run without modifications, can it? If this is an exact mirror or RPi.GPIO, the user expects a specific behaviour from 'pull_up_down', and that behaviour is not implemented. 

I think it atleast make sense to throw a warning stating that 'pull_up_down' is not supported on the Jetson platforms and that using 'pull_up_down' might lead to unexpected results.

swarren, 2020-11-23:
I was previously unsure how to implement a warning in a way that avoids warning every time that setup() is called. However, since the argument is optional, we can warn whenever it's passed, and not warn if the default is used. I'll push a change to implement this shortly.

maxvfischer, 2020-11-24:
@swarren Thanks, I think that make sense!

kylemcdonald, 2021-01-25:
just spent hours on this 😭 glad it has been turned into a warning in new versions.