NVIDIA/jetson-gpio #6: this GPIO library is too easy,we need linke raspberry Full library

zouzhe1, 2019-06-06:
now,the gpio lib only have gpio
i want to do pwm,i2c,spi,but i can`t see it
how long it can realize?
it best to like the Arduino.
i am Jetson Nano

swarren, 2019-06-06:
The intent of this library is to be a compatible re-implementation of the existing library RPi.GPIO, to allow existing code to run on Jetson.

RPi.GPIO doesn't have I2C or SPI APIs, and so Jetson.GPIO will not either. There are existing user-space APIs for I2C and SPI. See for example:
* Source for https://linux.die.net/man/8/i2cdetect
* https://www.kernel.org/doc/Documentation/i2c/dev-interface
* https://www.kernel.org/doc/Documentation/spi/spidev

PWM support does exist in RPi.GPIO but not yet in Jetson.GPIO. We are investigating whether we can add PWM support.

swarren, 2019-06-17:
PWM support is now implemented in v1.0.0. I2C and SPI are out of scope. Hence, closing.

closed 2019-06-17 by swarren