NVIDIA/jetson-gpio #8: Nano: No .input, no .PWM

vectorkt, 2019-07-05:
Neither .input nor .PWM work on the Jetson Nano with GPIO.BOARD configuration. Identical code & connected circuitry works as expected on the rasberry pi.

swarren, 2019-07-05:
Which version of the code are you using? With the latest code, I have personally tested all GPIOs on all Jetson boards in output mode, and all PWMs, and a subset of GPIOs for output (so I know that output functionality works in general, and output mode uses the same pin mapping as input). Please post the test code you're using, or try using the sample or test applications included in this repository.

P.S. I'm on vacation for a week or so, so won't be able to respond for a while.

swarren, 2019-07-18:
I assume that any issues are resolved by using the latest code in this repo, hence closing. If the issue still remains, please re-open and provide your test code and details of what hardware you're connecting. Thanks.

closed 2019-07-18 by swarren

hkk0x26, 2019-07-26:
I'm trying to run simple_pwm.py code but .pwm is not working, do we have to change the pinmux configuration to make it work.