NVIDIA/jetson-gpio #10: button_event.py does not always work

mpena2099, 2019-07-17:
Hi!

I'm testing the GPIO with button_event.py example. Sometimes when I press the button the event "button pressed" is activated. Sometimes that does not happen (I need to wait and try again). I'm using a 4.7k resistor as pull-up.

Thanks!
Mauricio

swarren, 2019-07-17:
Instead of using a button and a pull resistor, can you connect a wire to the input pin, and then connect the other end of that wire to GND and 3V3 alternatively (not at the same time) and see if that helps? I wonder if you're seeing the same issue that's described at https://devtalk.nvidia.com/default/topic/1052582/jetson-nano/jetson-gpio-interrupt-processing-on-nano/post/5357289/#5357289 ?

mpena2099, 2019-07-18:
@swarren, alternately connecting to GND and 3v3 manually seems to work! :open_mouth: 

Tested on pins 18 and 16. I'm note sure if it is the same problem as that one on Nvidia forum.

So the problem is... 🤔

swarren, 2019-07-18:
Could you please try a 1K pull resistor? The stronger pull might help.

mpena2099, 2019-07-18:
OK @swarren, now it works, thanks! :smile:

But I'm not sure why. There is already an active internal pull-up resistor on that pin or what? 

I think I would burn the IO pin on other boards if I connect like this (1k resistor or none), as the current limit is 1mA...

swarren, 2019-07-19:
This may be related to how the voltage level translators on Jetson work. We're investigating and hope to provide some guidance in the future.

Given that you've observed Jetson.GPIO working correctly now, and this issue appears hardware-related, are you OK closing this bug? Thanks.

mpena2099, 2019-07-19:
OK, thanks!

closed 2019-07-19 by mpena2099