NVIDIA/jetson-gpio #14: PWM pulses not reaching pin

hkk0x26, 2019-07-27:
```python

import RPi.GPIO as GPIO
import time

output_pins = {
    'JETSON_XAVIER': 18,
    'JETSON_NANO': 33,
}
out_pin = output_pins.get(GPIO.model, None)

def main():

    GPIO.setmode(GPIO.BOARD)
    print('board is set')
    GPIO.setup(out_pin, GPIO.OUT, initial=GPIO.HIGH)
    print('pin out set to high')
    p = GPIO.PWM(out_pin, 50)
    print('pwm pin set')
    p.start(0)
    print('pwm is started')
    
    try:
        while 1:
            for dc in range(0, 101, 5):
                print('+ve Duty cycle')
                p.ChangeDutyCycle(dc)
                time.sleep(0.1)
            for dc in range(100, -1, -5):
                print('-ve duty cycle')
                p.ChangeDutyCycle(dc)
                time.sleep(0.1)
    except KeyboardInterrupt:
        pass
    p.stop()
    GPIO.cleanup()
    print('pwm cycle completed')

if __name__ == '__main__':
    main()

```
Once while loop start executing motors stops running. It seems pwm signal is not reaching pin 33. using L298N pwm motor controller.

swarren, 2019-08-13:
You need to configure the relevant pins for PWM mode outside the Jetson.GPIO library. By default, all of the pins only work in GPIO mode. Search for "Jetson Nano Developer Kit 40-Pin Expansion Header Configuration New" at https://developer.nvidia.com/embedded/downloads for details re: how to configure the pins.

Alternatively, it is possible to cheat and dynamically change the pinmux configuration by directly writing to chip registers. This is not supported at all, but likely works in practice. You can find the cheat commands for some pins at https://github.com/NVIDIA/jetson-gpio/blob/master/samples/test_all_apis.py#L67

hkk0x26, 2019-08-14:
I manage to generate PMW signal for L298N using pca9685 with the help of Adafruit_Python_PCA9685 library. But still want to see pwm signal generating on nano board. Let me try above option.

swarren, 2019-08-23:
Can we close this bug now? Thanks.

hkk0x26, 2019-08-28:
Let me check the option provided by you and confirm the same, then we will close the issue

closed 2019-08-28 by hkk0x26

imadAlgDev, 2020-02-03:
hi,did u find a solution how to control dc motor from nano by l298N??