NVIDIA/jetson-gpio #15: test_all_apis.py fails for test_out_in_init_high

brtknr, 2019-07-28:
**Describe the bug**
A clear and concise description of what the bug is, including
- Which version of Jetson.GPIO you are using.
1.0.0
- Which version of L4T (not JetPack) you are using.
18.04.2
- Which Jetson board you are using.
Nano

```
Traceback (most recent call last):
  File "./test_all_apis.py", line 622, in <module>
    test()
  File "./test_all_apis.py", line 289, in test_out_in_init_high
    assert(val == GPIO.LOW)
AssertionError
```

**To Reproduce**
Steps to reproduce the behavior:
1. No pins wired to the board
2. Run `sudo ./run_sample.sh test_all_apis.py`

**Expected behavior**
For all tests to run successfully.

**Additional context**
Add any other context about the problem here.

swarren, 2019-08-13:
For this test script to run correctly, you must connect a wire between the pin the test uses for output and the pin the test uses for input. You can find the pin numbers in the test source:
https://github.com/NVIDIA/jetson-gpio/blob/master/samples/test_all_apis.py#L74

swarren, 2019-08-23:
Can we close this bug now? Thanks.

swarren, 2019-09-04:
Since this issue does not describe an actual bug in Jetson.GPIO, I am closing it.

closed 2019-09-04 by swarren