NVIDIA/jetson-gpio #17: getmode? what mode needs to be set to read from pin13 of Xavier GPIO [PWM01]/GPIO424

AndreV84, 2019-08-04:
`python3
Python 3.6.8 (default, Jan 14 2019, 11:02:34) 
[GCC 8.0.1 20180414 (experimental) [trunk revision 259383]] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import Jetson.GPIO as GPIO
>>> mode = GPIO.getmode()
>>> GPIO.JETSON_INFO
{'P1_REVISION': 1, 'RAM': '16384M', 'REVISION': 'Unknown', 'TYPE': 'Jetson Xavier', 'MANUFACTURER': 'NVIDIA', 'PROCESSOR': 'ARM Carmel'}
>>> `
```
>>> import Jetson.GPIO as GPIO
>>> GPIO.setmode(GPIO.TEGRA_SOC)
>>> mode = GPIO.getmode()
>>> GPIO.getmode()
1000

```

AndreV84, 2019-08-12:
which pin as per https://www.jetsonhacks.com/nvidia-jetson-nano-j41-header-pinout/ will be a pin for RPi 4 pin as in https://medium.com/dyi-electronics/raspberry-pi-and-dht11-humidity-sensor-cccf6b3072ad

swarren, 2019-08-13:
> Is there any knowledge about sensors that worked with the Jetson.GPIO approach?

Any sensor that is capable of being controlled using GPIOs should work fine.

> Any insights on how to read from dht11 that driver is integrated into tegra kernel?

In order to access a device using an existing kernel driver, you would have to use that driver's own APIs and not Jetson.GPIO.

> which pin as per ... will be a pin for RPi 4 pin as in ...

If you set the mode to BOARD or to BCM, then Jetson.GPIO will use the exact same pin numbers as the Raspberry Pi.

AndreV84, 2019-08-14:
Hi, thank you for your response;
Could you point out which will be pin4 RPi mapped to nano?
I  am trying to get dht11 to work with any Jetson but it will fali due to complexity

swarren, 2019-08-14:
In BOARD mode, on both the Raspberry Pi and on Jetson, pin 4 on the connector is pin 4 in software.

Sorry, but I don't have any reference software for any particular sensor; the job of Jetson.GPIO is simply to provide raw access to GPIOs. Any higher level protocol or sensor interaction is beyond its scope.

swarren, 2019-08-23:
Can we close this bug now? Thanks.

swarren, 2019-09-04:
Since this issue does not describe an actual bug in Jetson.GPIO, I am closing it.

closed 2019-09-04 by swarren