NVIDIA/jetson-gpio #19: Exception: Could not determine Jetson model

terryfitz, 2019-08-24:
Jetson.GPIO 2.0.0
Running on Raspberry Pi 3B+ (machine is armvl)
Kernel: 4.14.98-v7+

Installation of adafruit's [adafruit-blinka](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi) has a Jetson.GPIO dependency which was downloaded and installed using pip3.

Error returns 'Could not determine Jetson model' from gpio_pin_data.py.

My /proc/device-tree/compatible looks like:
raspberrypi,3-model-b-plus^@brcm,bcm2837^@

Question: Does jetson-gpio not support raspberry pi hardware? Can I manually modify my compatible file to include a compatible board, and if so which one? Thanks

swarren, 2019-08-24:
@terryfitz what version of pip do you have (run `pip --version`), and are you using a Python virtual environment or not (if so, please activate it before querying the pip version). Thanks.

swarren, 2019-08-24:
Also, could you please run `uname -a`.

(as background re: pip, IIRC older versions of pip don't honor the platform qualifications in a package's dependency file, so that might be why the Jetson library is getting installed on a Pi. Another issue might be if the system thinks it's aarch64 not armvl). Thanks.

terryfitz, 2019-08-25:
OP updated accordingly

terryfitz, 2019-08-25:
I walked through the trace and it was just odd that the __init__.py in RPi/GPIO was importing Jetson.GPIO. I double checked the [source](https://pypi.org/project/RPi.GPIO/) and there wasn't a match despite my system telling me that the RPi.GPIO version was the same as the source. So uninstalled RPi.GPIO and reinstalled and it works!

Thanks for the tip regarding different platform supports!

Now for some motor tinkering >_>

closed 2019-08-25 by terryfitz

terryfitz, 2019-08-29:
I downloaded the image of 2019-06-19-ubiquity-xenial-lxde from ubiquity robotics from https://downloads.ubiquityrobotics.com/pi.html
After installation I ran
`sudo apt-get upgrade`
`sudo apt-get update`

Based on your prompt I activated a virtual environment and 
```
(venv3.5_ros) ubuntu@ubiquityrobot:~/rospy3_kinetic$ pip install adafruit-blinka
Collecting adafruit-blinka
Collecting sysv-ipc (from adafruit-blinka)
Collecting RPi.GPIO (from adafruit-blinka)
Collecting Adafruit-PlatformDetect (from adafruit-blinka)
Collecting Adafruit-PureIO (from adafruit-blinka)
Collecting rpi-ws281x>=4.0.0 (from adafruit-blinka)
  Using cached https://files.pythonhosted.org/packages/32/99/1140d72d58e31d1251c2009238473e818f2d991e130c59f09dc7502d99bc/rpi_ws281x-4.2.2-cp35-cp35m-linux_armv7l.whl
Collecting Jetson.GPIO (from adafruit-blinka)
Collecting spidev (from adafruit-blinka)
  Downloading https://files.pythonhosted.org/packages/fb/14/4c2e1640f0cb04862c76d9d76ed7c945b0f67876e503ac02f7f675fe86a0/spidev-3.4.tar.gz
Building wheels for collected packages: spidev
  Building wheel for spidev (setup.py) ... done
  Created wheel for spidev: filename=spidev-3.4-cp35-cp35m-linux_armv7l.whl size=31885 sha256=abeaadb2bcfd283e9ebd0f7510eb346a22acb2764d04746e8ac7eb2294da3b93
  Stored in directory: /home/ubuntu/.cache/pip/wheels/10/d6/98/ba1f1999099e3e7adb3a0140af8c9287ff2b067ff005485674
Successfully built spidev
Installing collected packages: sysv-ipc, RPi.GPIO, Adafruit-PlatformDetect, Adafruit-PureIO, rpi-ws281x, Jetson.GPIO, spidev, adafruit-blinka
Successfully installed Adafruit-PlatformDetect-1.3.2 Adafruit-PureIO-0.2.3 Jetson.GPIO-2.0.0 RPi.GPIO-0.7.0 adafruit-blinka-2.3.2 rpi-ws281x-4.2.2 spidev-3.4 sysv-ipc-1.0.0
```

I checked `$ nano venv3.5_ros/lib/python3.5/site-packages/RPi/GPIO/__init__.py`

which returns
```
from Jetson.GPIO import *
VERSION = '0.1.0'
```

I didn't run the same blinkatest.py file as before but I'm sure that the same error will occur as RPi.GPIO is handing the torch to JETSO.GPIO

swarren, 2019-08-29:
Note that this is actually a bug in Adafruit Blinka's setup.py, or pip's processing of it. For some reason, it is installing Jetson.GPIO but shouldn't be doing so on your system. I'll try and track down what's triggering that, but need a Pi to do so. I have one at home but not at work, so it might be a few days.

swarren, 2019-08-31:
I used that Ubiquity image on a Pi 3B+ and still don't see Jetson.GPIO being installed automatically when adafruit-blinka is installed using pip. I've run out of things to try to reproduce your issue.