NVIDIA/jetson-gpio #20: No file 99-gpio.rules

VincentDuf, 2019-09-04:
Hi,
I just followed the installation instruction to install Jetson.GPIO and at the Permission change step I must copy the 99-gpio.rules file to the rules.d directory...but I can't find the 99-gpio.rules...

I'm using virtualenv and I installed it using pip, is that an issue ? 
Thanks

swarren, 2019-09-04:
Oh, this bug report has been edited. It originally reported that file 99-gpio.rules wasn't included in a package that had been installed by pip. This has been fixed by release 2.0.1

swarren, 2019-09-04:
For your new issue, can you please run:
```
id -a
ls -l /sys/class/gpio/export
ls -l /sys/class/gpio/unexport
```

(Note: you will need to log out/in to pick up the changes made by the `groupadd`/`usermod` commands; did you do that?)

VincentDuf, 2019-09-05:
Hi,
here is what I get from the command you asked:
```
nvidia@tegra-ubuntu:~$ id -a
uid=1001(nvidia) gid=1001(nvidia) groups=1001(nvidia),4(adm),20(dialout),24(cdrom),25(floppy),27(sudo),29(audio),30(dip),44(video),999(gpio)
nvidia@tegra-ubuntu:~$ ls -l /sys/class/gpio/export 
--w--w---- 1 root gpio 4096 Sep  5 07:39 /sys/class/gpio/export
nvidia@tegra-ubuntu:~$ ls -l /sys/class/gpio/unexport 
--w--w---- 1 root gpio 4096 Sep  5 07:39 /sys/class/gpio/unexport

```

I also see you modified the instruction but it's not clear in which depository you must be to find the 99-gpio.rules file...

If I install using pip in my virtualenv it should be " .virtualenv/name_virtual_env/lib/pythonNN/site-packages/Jetson/GPIO/" right ? Cause I don't have the file in this path...

swarren, 2019-09-05:
Great. Closing the bug then. Thanks.

closed 2019-09-05 by swarren

benjagracia, 2019-11-22:
I have the same problem, i am using a jetson nano

i tried running the commands several times and it didnt work

swarren, 2019-11-22:
@benjagracia the best place to get support for Jetson.GPIO is the NVIDIA forums:
https://devtalk.nvidia.com/default/board/139/embedded-systems/1

angora0606, 2023-03-23:
I have same problem and I can't solve this.Im using jetson nano.I tried Installation using pip and manual download but didn't change.Same problem keep going please help.

architjaiswal, 2023-04-30:
I'm also facing the same issue! My error says:  raise RuntimeError("The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that {} does not exist. Please check if that file is present.".format(_GPIOCHIP_ROOT))
RuntimeError: The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that /dev/gpiochip0 does not exist. Please check if that file is present.

AoiMatsu, 2023-05-03:
To solve this problem, run the following code in sequence:
sudo usermod -aG gpio $USER
sudo chown root.gpio /dev/gpiochip0
sudo chmod 660 /dev/gpiochip0

architjaiswal, 2023-05-03:
This solution works! Thank you

Hokaage, 2023-05-04:
@AoiMatsu 
This solution works but every time I reboot it resets the system and I have to rewrite

willertan, 2023-05-23:
> @AoiMatsu This solution works but every time I reboot it resets the system and I have to rewrite

I added this into crontab -e , it will wait 5 sec after boot and apply the permissions. it is ugly but works. make sure you are root when adding crontab  

@reboot sleep 5; usermod -aG gpio $USER; chown root.gpio /dev/gpiochip0; chmod 660 /dev/gpiochip0;

AoiMatsu, 2023-05-23:
I think you can follow the following steps to solve the problem of rewriting after reboot, but I didn't try:
①Open the `/etc/rc.local` file for editing:
sudo nano /etc/rc.local
②Add the following commands to the file (before the "exit 0" line):
usermod -aG gpio $USER
chown root.gpio /dev/gpiochip0
chmod 660 /dev/gpiochip0
③Save and exit (ctrl + x, then press y, then enter).

AoiMatsu, 2023-05-23:
@Hokaage @willertan I think you can follow the following steps to solve the problem of rewriting after reboot, I have tried and succeeded.(●'◡'●)

1. Create a new udev rules file, such as `/etc/udev/rules.d/99-gpio.rules`:

```bash
sudo nano /etc/udev/rules.d/99-gpio.rules
```

2. In this file, add the following line:

```text
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c '\
    chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
    chown -R root:gpio /dev/gpiochip0 && chmod -R 660 /dev/gpiochip0\
'"
```

3. Save and close the file.
4. Add the user to the gpio group, which can be done with the following command:

```bash
sudo usermod -aG gpio $USER
```
This command only needs to be executed once. After rebooting your device, the changes made will take effect.

5.reboot
```bash
reboot
```

I hope this can help you!

Qmditmore, 2023-05-23:
I have the issue, but none of these solutions have worked for me.

NaufalTsany, 2024-06-24:
hello @AoiMatsu, i tried yout solution but it didnt work for me it said error permission denied, my error says

`Traceback (most recent call last):
  File "/home/pal/venv/lib/python3.8/site-packages/Jetson/GPIO/gpio_cdev.py", line 147, in chip_open
    chip_fd = os.open(gpio_chip, os.O_RDONLY)
PermissionError: [Errno 13] Permission denied: '/dev/gpiochip1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pal/Documents/buzz.py", line 10, in <module>
    GPIO.setup(buzzer_pin, GPIO.OUT)
  File "/home/pal/venv/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 360, in setup
    _do_one_channel(ch_info, direction, init, consumer)
  File "/home/pal/venv/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 155, in _do_one_channel
    ch_info.chip_fd = gpio_cdev.chip_open_by_label(ch_info.gpio_chip)
  File "/home/pal/venv/lib/python3.8/site-packages/Jetson/GPIO/gpio_cdev.py", line 184, in chip_open_by_label
    chip_fd = chip_check_info(label, gpio_device)
  File "/home/pal/venv/lib/python3.8/site-packages/Jetson/GPIO/gpio_cdev.py", line 158, in chip_check_info
    chip_fd = chip_open(gpio_device)
  File "/home/pal/venv/lib/python3.8/site-packages/Jetson/GPIO/gpio_cdev.py", line 149, in chip_open
    raise GPIOError(e.errno, "Opening GPIO chip: " + e.strerror)
Jetson.GPIO.gpio_cdev.GPIOError: [Errno 13] Opening GPIO chip: Permission denied`

i tried to find for another forum that discuss this error but i could not find anything, do you know any solution to this error?

Minim9090, 2024-08-23:
> To solve this problem, run the following code in sequence: sudo usermod -aG gpio $USER sudo chown root.gpio /dev/gpiochip0 sudo chmod 660 /dev/gpiochip0

yeah this works for me