NVIDIA/jetson-gpio #22: what are PWM software limitations?

JoseColombini, 2019-09-13:
Hello, I'm trying to use a servo SG90 with a Jetson nano, but without successes.

This servo go to 90 degrees with 10% of DC and -90 with 5%, but it always rotate to a maximum of one side, not the other. And after rotate it do not have any resistance to be move back to any position.

I'm using a simple copy of simple_pwm.py just changing the p.start()

My question are: can pwm using python be accurate enough to do this task and I have other problem than the python script? Is pwm sending continuous?

swarren, 2019-09-13:
Jetson.GPIO only implements hardware PWM, not software PWM.

Have you modified the hardware pinmux settings so that the PWM signal is actually routed to the pin on the expansion header? Without that, the PWM functionality won't work. Instructions to modify the pinmux are part of the L4T documentation on the L4T download website; https://developer.nvidia.com/embedded/downloads.

swarren, 2019-10-07:
Closing due to no response.

closed 2019-10-07 by swarren