NVIDIA/jetson-gpio #23: AttributeError from Jetson Nano GPIO in python.

dennySORA, 2019-10-05:
Jetson Nano GPIO in python.

This is error code:
```
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import Jetson.GPIO as GPIO
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py", line 70, in <module>
    model, JETSON_INFO, _channel_data_by_mode = gpio_pin_data.get_data()
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio_pin_data.py", line 286, in get_data
    revision = f.split('-')[-1]
AttributeError: '_io.TextIOWrapper' object has no attribute 'split'
```

I found this error "split" from .egg packages.

GKFuzzy, 2019-10-05:
got same error, had to downgrade to Jetson.GPIO==2.0.1, last vers is bugged

swarren, 2019-10-07:
This should be fixed by v2.0.3.

closed 2019-10-07 by swarren