NVIDIA/jetson-gpio #24: Jetson.GPIO, RPi.GPIO and python-spidev

32nguyen, 2019-10-07:
Hi, 

I am trying to control a device (Unicorn HAT HD) using Jetson Nano through SPI. Unicorn HAT HD has been commonly used with Raspberry Pi. In this case, the required python package **_python-spidev_** is somehow interfaced with RPi.GPIO. I wonder if I use Jetson.GPIO, what should I do to replace RPi.GPIO to avoid conflict? Or do I need to do that? 

Hope my questions make sense. 

Tom

swarren, 2019-10-07:
Jetson.GPIO is intended to be a drop-in replacement for RPi.GPIO, and can even be imported as RPi.GPIO. So, you should be able to simply install Jetson.GPIO, not install RPi.GPIO, and then it sounds like python-spidev should just work.

32nguyen, 2019-10-07:
Thank @swarren for your quick answer.

closed 2019-10-08 by swarren