NVIDIA/jetson-gpio #25: problem with GPIO.remove_event_detect()

z14git, 2019-10-09:
- Jetson.GPIO version:  '2.0.1', '2.0.3'
- L4T version: nvidia-l4t-core 32.2.1-20190812212815
- Jetson board: Nano

**To Reproduce**
Steps to reproduce the behavior:
1. run these code in jupyter lab
```
import Jetson.GPIO as GPIO
GPIO.setmode(GPIO.TEGRA_SOC)

in_pin = 'SPI2_MISO'
GPIO.setup(in_pin, GPIO.IN)

def cb_fn(channel):
    pass

GPIO.add_event_detect(in_pin, GPIO.RISING, callback=cb_fn, bouncetime=200)
```
2. make a FALLING Edge on `in_pin`
3. run the code below in jupyter lab, and the kernel will keep busy
```
GPIO.remove_event_detect(in_pin)
```
4. if interrupt the kernel, it will show:
```
---------------------------------------------------------------------------
KeyboardInterrupt                         Traceback (most recent call last)
<ipython-input-6-6287b623d9e5> in <module>
----> 1 GPIO.remove_event_detect(in_pin)

/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py in remove_event_detect(channel)
    501 def remove_event_detect(channel):
    502     ch_info = _channel_to_info(channel, need_gpio=True)
--> 503     event.remove_edge_detect(ch_info.gpio)
    504 
    505 

/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio_event.py in remove_edge_detect(gpio)
    134     _set_edge(gpio, NO_EDGE)
    135 
--> 136     _mutex.acquire()
    137     del _gpio_event_list[gpio]
    138     _mutex.release()

KeyboardInterrupt: 
```

**Expected behavior**
`GPIO.remove_event_detect(in_pin)` should run successfully.

swarren, 2019-10-09:
If you instead run the following script from a shell (outside Jupyter), do you see the same issue? I don't see the issue in this case; I'm working on installig Jupyter to see if I can reproduce the issue then. Note: For the sample to work, you'll need to connect a wire between GPIO expansion connector pins 31 and 32.

```
#!/usr/bin/env python
  
import Jetson.GPIO as GPIO
import time

out_pin = 32
in_pin = 31

print("Setting up; out 1")
GPIO.setmode(GPIO.BOARD)
GPIO.setup(out_pin, GPIO.OUT)
GPIO.setup(in_pin, GPIO.IN)
GPIO.output(out_pin, GPIO.HIGH)

print("Adding callback")
def cb_fn(channel):
    print("callback")
GPIO.add_event_detect(in_pin, GPIO.RISING, callback=cb_fn, bouncetime=200)

time.sleep(1)
print("out 0")
GPIO.output(out_pin, GPIO.LOW)
time.sleep(1)

print("Calling GPIO.remove_event_detect")
GPIO.remove_event_detect(in_pin)
print("Calling cleanup()")
GPIO.cleanup()
```

swarren, 2019-10-09:
I can't reproduce this in Jupyter either. Do you have a notebook available that can reproduce this? Does the timing of executing the chunks of code in the notebook affect the issue at all? How exactly are you causing the falling edge on the input pin; perhaps the timing or mechanism has some impact.

z14git, 2019-10-11:
I just found this problem is caused by my external circuit. The code works fine when i fix my external circuit.

closed 2019-10-11 by z14git