NVIDIA/jetson-gpio #26: PWM Permission issue

pjueon, 2019-10-21:
Hi, I'm using Jetson Nano(I installed jebot image version 0p3p2)

I've just downloaded the pakage using pip.
And I set up the user permissions following Readme.md and it works fine.
( I downloaded 99-gpio.rules from the repository.)

I can run almost every samples on the repository without using sudo,
but I can't run simple_pwm.py without using sudo. With sudo, it works fine.

I chcecked the source, and I found out that  when I run it without using sudo, it stucks in the loop in _export_pwm function. 

I checked the permission in /sys/devices/7000a000.pwm/pwm/pwmchip0/pwm2
duty_cycle, enable and period files' permissions are  -rw-r--r-- 
Because I don't have permission to write enable, the program stucks in loop
(while not os.access(enable_path, os.R_OK | os.W_OK))

I also checked the permission in /sys/devices/7000a000.pwm/pwmchip0 , 
export and unexport files' permissions are --w--w--- , which are set correctly.

Can you give some advice? 

**Please don't submit questions**
Please use this form to report specific bugs in Jetson.GPIO. All general questions or support requests should be directed at NVIDIA's forums; see:
https://developer.nvidia.com/embedded/community/support-resources

Please delete this section from your bug report.

**Describe the bug**
A clear and concise description of what the bug is, including
- Which version of Jetson.GPIO you are using.
- Which version of L4T (not JetPack) you are using.
- Which Jetson board you are using.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect  button/wire/LED/... XXX to the board
2. Configure pin ... to input/output/...
3. Execute API ...

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

pjueon, 2019-10-22:
I downloaded JetBot SD card image from here.  
(https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Software-Setup)  
  
Yes, I copied the 99-gpio.rules from the repository to /etc/udev/rules.d/   
and I can see  /etc/udev/rules.d/99-gpio.rules contains lines that you described.  

Also I did reboot too.  
 
When I tried 'ls -l /sys/devices/*.pwm/pwmchip*/' I got this:   
  
ls: cannot access '/sys/devices/*.pwm/pwmchip*/': No such file or directory
 
 
I tried 'ls -l /sys/devices/*.pwm/pwm/pwmchip*/' instead,  and I got this:  
  
/sys/devices/7000a000.pwm/pwm/pwmchip0/:
total 0
lrwxrwxrwx 1 root root    0 Oct 21 17:10 device -> ../../../7000a000.pwm
--w--w---- 1 root gpio 4096 Oct 21 17:00 export
-r--r--r-- 1 root root 4096 Oct 21 17:10 npwm
drwxr-xr-x 2 root root    0 Oct 21 17:10 power
lrwxrwxrwx 1 root root    0 Oct 21 17:00 subsystem -> ../../../../class/pwm
-rw-r--r-- 1 root root 4096 Oct 21 17:00 uevent
--w--w---- 1 root gpio 4096 Oct 21 17:00 unexport

/sys/devices/70110000.pwm/pwm/pwmchip4/:
total 0
lrwxrwxrwx 1 root root    0 Oct 21 17:10 device -> ../../../70110000.pwm
--w--w---- 1 root gpio 4096 Oct 21 17:00 export
-r--r--r-- 1 root root 4096 Oct 21 17:10 npwm
drwxr-xr-x 2 root root    0 Oct 21 17:10 power
lrwxrwxrwx 1 root root    0 Oct 21 17:00 subsystem -> ../../../../class/pwm
-rw-r--r-- 1 root root 4096 Oct 21 17:00 uevent
--w--w---- 1 root gpio 4096 Oct 21 17:00 unexport
  

And after I executed the pwm sample by python /opt/nvidia/jetson-gpio/samples/simple_pwm.py
I got this from 'ls -l /sys/devices/*.pwm/pwm/pwmchip*/'

/sys/devices/7000a000.pwm/pwm/pwmchip0/:
total 0
lrwxrwxrwx 1 root root    0 Oct 21 17:10 device -> ../../../7000a000.pwm
--w--w---- 1 root gpio 4096 Oct 21 17:16 export
-r--r--r-- 1 root root 4096 Oct 21 17:10 npwm
drwxr-xr-x 2 root root    0 Oct 21 17:10 power
drwxr-xr-x 3 root root    0 Oct 21 17:16 pwm2
lrwxrwxrwx 1 root root    0 Oct 21 17:10 subsystem -> ../../../../class/pwm
-rw-r--r-- 1 root root 4096 Oct 21 17:10 uevent
--w--w---- 1 root gpio 4096 Oct 21 17:16 unexport

/sys/devices/70110000.pwm/pwm/pwmchip4/:
total 0
lrwxrwxrwx 1 root root    0 Oct 21 17:10 device -> ../../../70110000.pwm
--w--w---- 1 root gpio 4096 Oct 21 17:00 export
-r--r--r-- 1 root root 4096 Oct 21 17:10 npwm
drwxr-xr-x 2 root root    0 Oct 21 17:10 power
lrwxrwxrwx 1 root root    0 Oct 21 17:10 subsystem -> ../../../../class/pwm
-rw-r--r-- 1 root root 4096 Oct 21 17:10 uevent
--w--w---- 1 root gpio 4096 Oct 21 17:00 unexport

Thanks.

swarren, 2019-10-22:
It looks like the udev scripts are working fine for the export an unexport files. Perhaps the other rule isn't working.

Can you please run this:

```
echo 0 > /sys/devices/7000a000.pwm/pwm/pwmchip0/export
ls -l /sys/devices/7000a000.pwm/pwm/pwmchip0/pwm0
echo 2 > /sys/devices/7000a000.pwm/pwm/pwmchip0/export
ls -l /sys/devices/7000a000.pwm/pwm/pwmchip0/pwm2
```

pjueon, 2019-10-23:
When I tried this,

echo 0 > /sys/devices/7000a000.pwm/pwm/pwmchip0/export

I got this.

bash: echo: write error: Device or resource busy

And when I tied this,

ls -l /sys/devices/7000a000.pwm/pwm/pwmchip0/pwm0

I got this.

ls: cannot access '/sys/devices/7000a000.pwm/pwm/pwmchip0/pwm0': No such file or directory

And when I tried this, 

echo 2 > /sys/devices/7000a000.pwm/pwm/pwmchip0/export 
ls -l /sys/devices/7000a000.pwm/pwm/pwmchip0/pwm2

I got this result.

total 0
-r--r--r-- 1 root root 4096 Oct 22 17:16 capture
-rw-r--r-- 1 root root 4096 Oct 22 17:16 capture_window_length
-rw-r--r-- 1 root root 4096 Oct 22 17:16 double_period
-rw-r--r-- 1 root root 4096 Oct 22 17:16 duty_cycle
-rw-r--r-- 1 root root 4096 Oct 22 17:16 enable
-rw-r--r-- 1 root root 4096 Oct 22 17:16 period
-rw-r--r-- 1 root root 4096 Oct 22 17:16 polarity
drwxr-xr-x 2 root root    0 Oct 22 17:16 power
-rw-r--r-- 1 root root 4096 Oct 22 17:16 ramp_time
-rw-r--r-- 1 root root 4096 Oct 22 17:16 uevent

swarren, 2019-10-23:
Hmm. So for the pwm2 case, it looks like the udev rules aren't running correctly for the individual PWM channel even though they are for the PWM controller itself. Are you able to debug udev and find out why?

pjueon, 2019-10-24:
Sorry. I know nothing about udev rules(how to write, how they work, etc). So I have no idea how to debug udev...

But I've noticed something.

In  99-gpio.rules file.

SUBSYSTEM=="pwm", KERNEL=="pwmchip*", ACTION=="add", \
    PROGRAM="/bin/bash -c 'chown root:gpio /sys%p/{,un}export; chmod 220 /sys%p/{,un}export"

isn't it supposed to be /{,un}export'" ? 

I think "  '   " might be missing. 

But like I said, I don't know how to write udev rules so it might be correct in that way.

swarren, 2019-10-24:
Yes, that quote is needed. I'm not sure how my testing succeeded without that! Please add in that quote, reboot, and see if that solves the issue for you. Thanks.

swarren, 2019-10-24:
I have confirmed that the udev rules work for me even with the missing quote. Besides, the missing quote is on the line that's working for you! I'll fix this, but I don't think it will help you. So, there must be something else going on.

swarren, 2019-10-25:
Aha. I tracked down a larger SD card so I could test with that image, and reproduced the problem. The issue is that image is based on L4T r32.1, but the udev rule for PWM channels relies on a kernel patch that was added in L4T r32.2. If you update the L4T release you're using (or rather, persuade the author of that SD card image to update their image), then everything should work.

pjueon, 2019-10-26:
Thank you very much for the information. 

Is it possible to update L4T version without completely re-flashing? 
If it's not, I think I should use PWM with sudo for now.

swarren, 2019-10-28:
> Is it possible to update L4T version without completely re-flashing?

It is not possible with current releases.

pjueon, 2019-10-29:
I see. Thank you very much for your help!

closed 2019-10-29 by swarren