NVIDIA/jetson-gpio #28: Now working with systemm service

ychen1986, 2019-10-28:
**Describe the bug**
Not working with systemd 

A clear and concise description of what the bug is, including
- Which version of Jetson.GPIO you are using.
{'P1_REVISION': 1, 'RAM': '8192M', 'REVISION': 'Unknown', 'TYPE': 'Jetson TX2', 'MANUFACTURER': 'NVIDIA', 'PROCESSOR': 'ARM A57 + Denver'}
- Which version of L4T (not JetPack) you are using.
NVIDIA Jetson TX2
 L4T 32.2.1 [ JetPack UNKNOWN ]
 Board: t186ref
 Ubuntu 18.04.3 LTS
 Kernel Version: 4.9.140-tegra
 CUDA 10.0.326
- Which Jetson board you are using.
Jetson TX2

**To Reproduce**
Steps to reproduce the behavior:
a. Creat a python script
 1.  Read the P12
 2. if is is low, sleep, otherwise shutdown the system
b.  add the script as a systemd service 
[Unit]
Description = This is a Service that shutdown the system when the P is loaded high. 
After=graphical.target

[Service]
#ExecStart=/home/nvidia/shutdown.sh
WorkingDirectory=/home/nvidia/
ExecStart=/usr/bin/python3 /home/nvidia/yolov3/shutdown_jutson.py
Restart=always
StandardOutput=inherit
StandardError=inherit
User=root
#Group=nvidia
#RestartSec=10s
Type=simple
[Install]
WantedBy=graphical.target
#WantedBy=multi-user.target

**Expected behavior**
It is expected to work on boot up.
However, it seems the GPIO is not working at the begging. 
If I reset the P12 such as GPIO.setup(12, GPIO.IN) or reboot the python script. 
it works normally as expected.

Here is my script 

import Jetson.GPIO as GPIO
import time
import os
# Pin Definitions
input_pin = 18  # BCM pin 18, BOARD pin 12

def main():
    TimeToDown = 6
    mytimer = 0
    time.sleep(10) 
    value = GPIO.LOW
    # Pin Setup:
    GPIO.setmode(GPIO.BCM)  # BCM pin-numbering scheme from Raspberry Pi
    GPIO.setup(input_pin, GPIO.IN)  # set pin as an input pin
    print("Starting demo now! Press CTRL+C to exit")
    try:
        while True:
#           value = GPIO.HIGH
            value = GPIO.input(input_pin)
            if mytimer != TimeToDown:
                if value != GPIO.HIGH:
                    if mytimer != 0:
                        print("Power supply recovered!")
                    mytimer = 0
                else:
                    print("{} seconds left to down".format(TimeToDown - mytimer))
                    mytimer = mytimer + 1
                    time.sleep(1)
            else:
                print("Shut OS down...")
                time.sleep(2)
                os.system('shutdown now')
    finally:
        GPIO.cleanup()

if __name__ == '__main__':
    main()

swarren, 2019-10-28:
What do the logs show for your service when it's not working? You can read them by running:
`journalctl -u name_of_your_service` or looking in various files in `/var/log`.

ychen1986, 2019-10-29:
I have addressed the problem.
It seems that the GPIO.setup() does not work correctly when booting, possibly due to some dependencies. 
To address the problem, I tried two approaches. 
One is to insert a 10 second delay before the GPIO.setmode and setup.
The other is to call GPIO.setup every time when call GPIO.input()
Both work as expected.

swarren, 2019-10-29:
Did you capture the actual error in the log files? If so, I'd be curious what the issue was.

swarren, 2019-12-02:
Closing issue since it appears stale.

closed 2019-12-02 by swarren