NVIDIA/jetson-gpio #32: GPIO warning not carrier board

navyjosh, 2020-02-25:
**Describe the bug**
- Which version of Jetson.GPIO you are using.
Jetson.GPIO (2.0.6)

- Which version of L4T (not JetPack) you are using.
# R32 (release), REVISION: 3.1, GCID: 18186506, BOARD: t210ref, EABI: aarch64, DATE: Tue Dec 10 06:58:34 UTC 2019

**To Reproduce**
1. Install OS using jetpack (target device jetson nano developer kit)
2. sudo pip3 install jetson.GPIO
3. import jetson.GPIO

**Expected behavior**
I expect GPIO pins to function, but they are not functioning

**Additional context**

--not sure if relevant-- install cuda, deepstream, various libs required for jetson-inference using apt.
https://github.com/dusty-nv/jetson-inference

For jetson-inference I followed instructions to build from source.
https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md

swarren, 2020-02-25:
For some reason, the software does not detect the Jetson Nano Developer Kit's baseboard. Have you ever unplugged the module from the baseboard? Either way, could you make sure the module and baseboard are fully plugged together.

Do you have a connection to the debug serial port? If so, if you could collect the full boot logs (i.e. including the bootloader output, not just the kernel's dmesg output) that may be useful.

Since this sounds like a hardware issue, you may get better support from NVIDIA's developer forums (https://devtalk.nvidia.com/default/board/139/embedded-systems/1); it sounds like the problem isn't related to Jetson.GPIO itself, but rather is something wider.

navyjosh, 2020-02-25:
I don't have a baseboard, does that mean I have the jetson nano and _not_ the jetson nano developer kit?

On the box it says developer kit, but it only came with one device.  There was no module + baseboard.

navyjosh, 2020-02-25:
Okay then to answer your question, no, I have never removed the module from the carrier board.

navyjosh, 2020-02-25:
The picture you showed has two slots for MIPI camera.  My board has only one slot for MIPI.  Could my board be out-dated?

swarren, 2020-02-25:
Oh, that comment posted? Github was telling me it couldn't post it. Oh well.

There are a couple of different versions of the baseboard. One has just 1 CSI connector, and the other has 2. Both should work fine with Jetson.GPIO, and both should be detected and reported in `/proc/device-tree/chosen/plugin-manager/ids`.

swarren, 2020-02-27:
Were you able to resolve this issue, or determine whether it's a hardware problem, e.g. try the same software in a different Jetson system and see if you still have the problem?

ryanaltair, 2020-03-07:
some suggustion, to make pins work, for example, input pins need a pull-up  
[image](https://github.com/ryanaltair/jetson_gpio_sh/blob/master/doc/read_D18.svg)

navyjosh, 2020-03-21:
Hello, I think the GPIO hardware was damaged due to exposure to a +5V external power on the +5V output pin on the GPIO.  Perhaps the error message should also include the possibility for hardware damage.

closed 2020-03-21 by navyjosh

BaptisteAmato, 2020-08-13:
Hi, I'm having the same issue with my Jetson Nano (developer kit), however the pins work correctly when looking at my multimeter (when connecting the 5V pin to the 18th pin for example, I do get 5v on the 18th, but not detected by the lib...).

I'm also using an old version of Jetpack (3.1).

swarren, 2020-08-13:
Baptiste, please file a separate bug since this bug is closed (the issue was HW damage and not a problem in Jetson.GPIO), and you have a different symptom too.

When you file the bug, please run the command "ls -l /proc/device-tree/chosen/plugin-manager/ids" and include the output in your bug report. Thanks.