NVIDIA/jetson-gpio #33: Permission error after update 2.0.6

Visuatech, 2020-02-28:
After the last update I get permission errors in my dockerized builds:

Traceback (most recent call last):
  File "./listen_input.py", line 2, in <module>
    import RPi.GPIO as GPIO
  File "/usr/local/lib/python2.7/site-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/usr/local/lib/python2.7/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/usr/local/lib/python2.7/site-packages/Jetson/GPIO/gpio.py", line 33, in <module>
    raise RuntimeError("The current user does not have permissions set to "
RuntimeError: The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this

I also tried this with python3, but also the same error there:
Traceback (most recent call last):
  File "./listen_input.py", line 2, in <module>
    import RPi.GPIO as GPIO
  File "/usr/local/lib/python3.8/site-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/usr/local/lib/python3.8/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/usr/local/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 33, in <module>
    raise RuntimeError("The current user does not have permissions set to "
RuntimeError: The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this

This used to work ~2 weeks ago without issue.

I ran the 
sudo groupadd -f -r gpio
sudo usermod -a -G gpio your_user_name
and copied the 99-gpio.rules and rebooted several times and rebuilt the image several times.

id -a
uid=1000(edgeai) gid=1000(edgeai) groups=1000(edgeai),4(adm),24(cdrom),27(sudo),29(audio),30(dip),44(video),46(plugdev),108(i2c),118(lpadmin),124(gdm),127(docker),130(sambashare),131(iotedge),998(weston-launch),999(gpio)

ls -l /sys/class/gpio/export
--w--w---- 1 root gpio 4096 feb 28 10:00 /sys/class/gpio/export

ls -l /sys/class/gpio/unexport
--w--w---- 1 root gpio 4096 feb 28 10:00 /sys/class/gpio/unexport

Any ideas?

swarren, 2020-02-28:
I don't *think* anything has changed recently in this permission checking code. What version did you have installed before, and how are you installing it (git, PyPi, ...).

Visuatech, 2020-03-02:
My mistake, I didn't run it in privileged mode.

closed 2020-03-02 by Visuatech