NVIDIA/jetson-gpio #36: No "99-gpio. rules file in the GPIO directory

TCIII, 2020-04-05:
**Please don't submit questions**
Please use this form to report specific bugs in Jetson.GPIO. All general questions or support requests should be directed at NVIDIA's forums; see:
https://developer.nvidia.com/embedded/community/support-resources

Please delete this section from your bug report.

**Describe the bug**
A clear and concise description of what the bug is, including
- Which version of Jetson.GPIO you are using.
- Which version of L4T (not JetPack) you are using.
- Which Jetson board you are using.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect  button/wire/LED/... XXX to the board
2. Configure pin ... to input/output/...
3. Execute API ...

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

TCIII, 2020-04-05:
Sorry, my issue is with JetPack and not L4T.

closed 2020-04-05 by TCIII

jorgemia, 2020-06-25:
@TCIII I can't find a rules file either. Were you able to fix it?

TCIII, 2020-06-25:
@jorgemia,

I ignored the warning and all was well.

Regards,
TCIII