NVIDIA/jetson-gpio #37: PWM not working on Jetson Xavier NX

ethancherki, 2020-05-19:
I am trying to use pwm on the GPIO 33 of the Xavier NX board but the sample script is not working. I have seen that I might have to configure the pinmux. However, there is a document called "Jetson Nano Developer Kit 40-Pin Expansion Header Configuration"  for the Nano but there is nothing for the Xavier NX. 

What should I do ? How can I configure the board properly ?

swarren, 2020-05-19:
Generally the same techniques apply to configuring the pinmux for any Jetson board.
- Use the Jetson.IO tool. This is the easiest method.
- Use the spreadsheet as described in that document. This is the preferred method if you're shipping a product based on Jetson.

ethancherki, 2020-05-26:
Thank you for your answer. PWM is now activated on PIN 33 and seems to be working. However when I try to increase the pwm frequency from 50 Hz to 100Hz or 1000 Hz, I get the following error:

File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 245, in _set_pwm_period
    f.write(str(period_ns))
OSError: [Errno 22] Invalid argument

Do you have any idea on how to solve this issue ? Is there any pwm frequency limitation on the Xavier NX ?

swarren, 2020-05-26:
The various boards and SW releases do have some different limitations on PWM period. If this isn't covered in the L4T documentation, the best place to get an answer is on the NVIDIA Jetson forums; https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/70. Jetson-gpio simply exposes whatever the kernel/hardware support without additional restriction.

swarren, 2020-06-08:
Closing; I assume that followup on the forum was successful.