NVIDIA/jetson-gpio #39: Wrong Linux GPIO numbering

Visuatech, 2020-06-05:
Hi!
I consulted your GPIO pin data for the NX to find out the linux pin number for board pin 31 after moving from a nano to an NX, and found that it might be wrong?
in the line:    
`(134, "/sys/devices/2200000.gpio", 31, 6, 'GPIO11', 'SOC_GPIO42', None, None),`
134 should be the linux GPIO pin number according to the file, but actually for GPIO11 the pin number is 388 (which I tested and is indeed correct) am I looking at it wrong?

for the nano the line is correct, as 200 is indeed the correct pin for GPIO11

swarren, 2020-06-05:
I believe that 134 is correct. SoC pin SOC_GPIO42 is GPIO Q.6, which is GPIO ID 134 according to both the Tegra194 pinctrl driver and the Tegra194 GPIO DT bindings header file. How did you derive the value 388? I've also tested all the pins for both input and output.

Visuatech, 2020-06-05:
I tried it with echo 134 > /sys/class/gpio/export and got consistent error:
`-bash: echo: write error: Invalid argument`
So I did a lot more googling and found this thread:
https://forums.developer.nvidia.com/t/gpio-doesnt-work/49203/14

The thread is about the TX2, but it's very strange since these were the only ones that work and actually correlate to the correct board pin
In that thread it's explaned that these GPIO's should be used:
All the GPIOs on J21:

GPIO8_ALS_PROX_INT = 388
GPIO9_MOTION_INT = 298
GPIO11_AP_WAKE_BT = 389
GPIO16_MDM_WAKE_AP = 481
GPIO19_AUD_RST = 398

So I tried 
`echo 388 > /sys/class/gpio/export`
followed by
`echo in > /sys/class/gpio/gpio388/direction`

And when I read the value using
`cat /sys/class/gpio/gpio388/value`
after connecting the pin to GND I can see the value change
I am using a Xavier NX development kit.

This might be caused by some other unrelated issue in which case I'll close the issue and post on the nvidia forums

swarren, 2020-06-08:
The assignments of SoC/module GPIOs to pins on the GPIO expansion connector is potentially different for each board, and the numbering of SoC GPIOs is different for each SoC. So, a thread about Jetson TX2 has no relevance to Jetson NX.

Are you using the Jetson.GPIO library, or directly accessing GPIOs yourself using sysfs? Please note that if you're doing the latter, the GPIO numbers in the Jetson.GPIO data tables are the GPIO number *within* a GPIO controller, whereas sysfs used global Linux GPIO numbers, so you need to add the gpiochip-specific offset. However, direct use of sysfs is out-of-scope for a bug report on Jetson.GPIO; the entire point of Jetson.GPIO is to abstract all of this so you don't care.

swarren, 2020-09-21:
I assume this issue is resolved now; please feel free to re-open it if the original question still remains unsolved. Thanks.

closed 2020-09-21 by swarren

paroque28, 2024-07-08:
Would be nice if Nvidia published the right mapping from Pin number to /sys/class/gpio/ ...

anhmiuhv, 2024-07-08:
We no longer use the sysfs approach in the latest version of kernel