NVIDIA/jetson-gpio #40: Event triggered twice despite high debounce time

tasiogr, 2020-07-02:
I'm playing with the button_interrupt.py sample and I'm getting some strange behavior.

I'm running Jetson.GPIO 2.0.8 on L4T # R32 (release), REVISION: 4.2, GCID: 20074772, BOARD: t210ref. Jetson Nano Development Kit B01.

I have connected a button to the J41 pin 31 (which is supposed to be by default configured as pull up, correct me if I'm wrong here). The pin is configured as input as per the sample.

If I use a not-so-low debounce time (even up to a second), I get an event triggered twice. If I raise the debounce time up to a point where I don't get the event repeated, the button stops responding and the python program will need two Ctrl-C until it gets the keyboard interrupt. I guess this indicates the thread gets stuck somewhere.

swarren, 2020-07-07:
Thanks for the report. Unfortunately, this will have to wait a while, since I don't have physical access to any Jetson in order to wire up some input to test this. I should have connected some loopback wires before I left the office.

Nisces, 2020-07-17:
I have the same problem, when set a bounce time(no matter what value), the program stops responding and need two Ctrl+C until it gets the keyboard interrupt.

swarren, 2020-08-11:
I have pushed version 2.0.9 which should resolve the hang and double-CTRL-C-required issues.

A terminology note: the word "debounce" in the event API doesn't mean what you might typically think debounce means. It's implemented as a simple maximum-callback-rate, rather than the likely expected don't-notify-until-the-value-has-been-stable-for-x-ms. This behaviour matches the original RPi.GPIO library, which Jetson.GPIO emulates in order to ensure application compatibility.

closed 2020-08-11 by swarren