NVIDIA/jetson-gpio #41: Insufficient permissions for upstart

Patrick-Thomas, 2020-09-01:
Hello, I am trying to run ROS automatically when booting up my Xavier NX using the robot_upstart package. The ROS core and most of the nodes get up and running successfully however when trying to load the Jetson-GPIO library I get the following error:

`File "/../../../button_input.py", line 6import Jetson.GPIO as GPIO
File "/usr/lib/python2.7/dist-packages/Jetson/GPIO/__init__.py", line 1, in <module>
from .gpio import *
File "/usr/lib/python2.7/dist-packages/Jetson/GPIO/gpio.py", line 33, in <module>
raise RuntimeError("The current user does not have permissions set to "
RuntimeError: The current user does not have permissions set to access the library function`

It looks like 99-gpio.rules isn't granting gpio permission to the daemon process started by robot_upstart. I've tried adding the following line to the bottom of 99-gpio.rules:

`ACTION=="add", KERNEL=="gpio", MODE="0666"`

but this hasn't made a difference. Do you know how I can make the gpio/pwm interfaces accessible to the daemon?

Thanks

swarren, 2020-09-01:
Whichever user your daemon runs as needs to be a member of the gpio group.

Another issue could be that your daemon is starting before all relevant filesystems are mounted and udev events have been proessed. Make sure the relevant systemd unit file has the correct dependencies specified.

Patrick-Thomas, 2020-09-01:
Thanks for your help. In the end I fixed this by modified a file generated in robot_upstart rather than 99-gpio.rules.

For anyone interested I recommend looking here: https://github.com/clearpathrobotics/robot_upstart/issues/18

closed 2020-09-01 by Patrick-Thomas