NVIDIA/jetson-gpio #42: simple pwm example script not working

skywo1f, 2020-09-10:
Add any other context about the problem here.
/jetson-gpio/samples$ python3 simple_pwm.py 
/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py:352: RuntimeWarning: This channel is already in use, continuing anyway. Use GPIO.setwarnings(False) to disable warnings
  RuntimeWarning)
/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py:593: RuntimeWarning: This channel is already in use, continuing anyway. Use GPIO.setwarnings(False) to disable warnings
  RuntimeWarning)
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "simple_pwm.py", line 61, in <module>
    main()
  File "simple_pwm.py", line 41, in main
    p = GPIO.PWM(output_pin, 13)
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 597, in __init__
    self._reconfigure(frequency_hz, 50.0)
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 635, in _reconfigure
    _set_pwm_period(self._ch_info, self._period_ns)
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 245, in _set_pwm_period
    f.write(str(period_ns))
OSError: [Errno 22] Invalid argument

swarren, 2020-09-10:
Can you please let me know:
* Which version of Jetson.GPIO you're using.
* Which L4T release you're using.
* Which hardware (Jetson board) you're using.

skywo1f, 2020-09-10:
I just downloaded Jetson.GPIO today, so the most recent one. Using NVIDIA L4T 32.4.3. This is on a Jetson Xavier devkit.

swarren, 2020-09-10:
I just run the latest Jetson.GPIO on that L4T release on Jetson Xavier without issue.

If you reboot, does the problem go away? The error message "RuntimeWarning: This channel is already in use, continuing anyway. Use GPIO.setwarnings(False) to disable warnings" indicates that there was some kind of problem cleaning up from a previous attempt to use PWM. I don't see that. I suspect this is at least related to the later "Invalid argument" error.

So, if you reboot and then run simple_pwm, which errors do you see, and if you then run simple_pwm a second time do you get a different set of errors?

swarren, 2020-09-21:
Does this problem still exist? Thanks.

skywo1f, 2020-09-29:
still did not work after reboot. I also tried the simple_out script. that one did not error, but it also did not do anything.

swarren, 2020-09-29:
Let's concentrate on getting simple_out working first.

* Did you modify the simple_out sample in any way? If so, please attach the exact code you ran.
* How did you determine whether this sample was working; did you attach a voltmeter, logic analyzer, oscilloscope, LED, or something else to Jetson? If a LED or something else, please attach a schematic or photo of it.
* Can you attach a photo showing which pin you connected your test device to?

Thanks.

swarren, 2020-10-08:
@skywo1f do you still have this issue? If so, could you please answer the questions in my previous comment. Thanks.

loublock, 2020-10-12:
Hello, I have the same problem.

I just installed the latest Jetson GPIO version (2.0.11) on my Jetson Xavier (L4T 32.3.1).
When executing the simple_pwm.py file I get the exact same error message. I'm starting further investigations.

loublock, 2020-10-12:
In your code (gpio.py) in the constructor it says _frequency_hz_:

`class PWM(object):
    def __init__(self, channel, frequency_hz):`

but I think you assume a cycle period in your sample_pwm.py, because the error clearly says:

__set_pwm_period(self._ch_info, self._period_ns)
File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 245, in _set_pwm_period_

So I tried a more realistic PWM frequency of 1 kHz:

`GPIO.PWM(18, 1000)`

and it worked like a charm.

swarren, 2020-10-12:
Following @loublock 's comment, I took a looked at the code and compared it to the exception backtrace in the original bug report from @skywo1f . The code in the bug report doesn't match the latest code. I believe the original problem has already been fixed. Specifically, the following code in the backtrace:

```File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 597, in init
self._reconfigure(frequency_hz, 50.0)
```

... is not present in the latest code, nor the latest code present when this bug was filed (the 50.0 is 0.0 now). It was removed in March 2020, and that change was included in Jetson.GPIO 2.0.7. I suspect you (@skywo1f) simply need to update to the latest version of Jetson.GPIO; you're probably running the version that came with L4T rather than the version you downloaded more recently.

I'll look into @loublock 's issue in a second.

swarren, 2020-10-12:
@loublock 

The parameters to the PWM API is definitely a frequency and not a period. This appears to be passed through the implementation correctly. The fact that the exception mentioned set_pwm_period is simply a coincidence, not because of a frequency/period mismatch anywhere in the code. Again, the most likely cause is that the code used to set the initial PWM duty cycle to 50%, which the kernel disallows under some circumstances. Jetson.GPIO  was fixed to set an initial duty cycle of 0% in the following commit, which should resolve this issue:

https://github.com/NVIDIA/jetson-gpio/commit/ac755d1ef297c764fcbc03c446b441fd7b2c01ae
(Part of Jetson.GPIO v2.0.7)

You probably need to install an updated version of Jetson.GPIO using pip, and preferably in a virtual environment to avoid interfering with L4T's system-wide packaging of Jetson.GPIO.

The sample uses a 50Hz PWM frequency because this is the typical frequency used by servo motors, which are a common use-case for PWM. Certainly other frequencies are useful depending on the use-case, but 50Hz should work fine too.

swarren, 2020-10-12:
BTW, you can query the version by:

```import Jetson.GPIO
print(Jetson.GPIO.VERSION)
```

loublock, 2020-10-14:
Dear Swarren, thank you for your response.
I've double checked on the Jetson.GPIO version I have installed. I also created a virtual env and installed the latest Jetson.GPIO version.

If I get you right, this error is related to the initial 50% duty cycle, which you fixed in 2.0.7. But I still have this error in 2.0.11, what do I miss?

loublock, 2020-10-14:
I played around a bit and found an issue trying to change the duty cycle or frequency: 

>>> `p.ChangeDutyCycle(50)`

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/zot/virtualenvs/PCAPS/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 632, in ChangeDutyCycle
    self._reconfigure(self._frequency_hz, duty_cycle_percent)
  File "/home/zot/virtualenvs/PCAPS/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 650, in _reconfigure
    _set_pwm_period(self._ch_info, self._period_ns)
  File "/home/zot/virtualenvs/PCAPS/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 244, in _set_pwm_period
    with open(_pwm_period_path(ch_info), 'w') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/32c0000.pwm/pwm/pwmchip2/pwm0/period'

In my system there is no _pwm0_ dir in _pwmchip2_. I have no clue why and how this is related to the other issue.

swarren, 2020-10-14:
@loublock

a) Was your original problem a failure in the code `p = GPIO.PWM(output_pin, 13)`? If not, it's a different issue; please open a different bug for that. This bug covers a problem creating the PWM object in the first place.

b) If you did originally have a problem calling `GPIO.PWM()`, but it's now fixed, how did you fix it? I assume you must have done, since you can't call `ChangeDutyCycle()` withoug `GPIO.PWM()` having succeeded.

Thanks.

loublock, 2020-10-15:
Hey Warren, I think this is a little bit messy right now.  

a) I wasn't able to create the PWM object with 50 Hz due to limitation in the kernel dtb as shown in this thread:
[https://forums.developer.nvidia.com/t/pwm-err-on-xavier-l4t-r32-3-1/109980](url)

b) Creating it with f >= 200 Hz works fine. Starting it with a duty cycle of 50 % also works fine, scope proofs. The error above occurs only when I try to change the frequency to 50 Hz after start. Because this fails, the object is somehow corrupt and other instructions are no longer possible, as I found out. 

So to clear things up, the original problem in this thread has nothing to do with mine and you can close this case. When I'm back in the lab I will finish my code to see if there are any other bugs remaining. 

Thank you for your help!

swarren, 2020-10-15:
@loublock thanks for the update.

@skywo1f  can you please let me know if you still have this issue? If so, please see https://github.com/NVIDIA/jetson-gpio/issues/42#issuecomment-701000805. I expect to close this bug in a couple of days if I don't hear back. Thanks.

swarren, 2020-10-23:
Closing stale bug.

closed 2020-10-23 by swarren