NVIDIA/jetson-gpio #44: Xavier NX, some pins do not work well

guizilaile23, 2020-09-21:
Hey guys, i test the Jetson.GPIO in python3, my code is realy very simple.

device XavierNX 
Jetpack : 4.4

> 
    import Jetson.GPIO as GPIO 
    import time
    GPIO.setmode(GPIO.BOARD)
    print(GPIO.getmode())
    channel = 13 
    GPIO.setup(channel, GPIO.OUT)
    while True:
        print('setting pin 13 high')
        GPIO.output(channel, GPIO.HIGH)
       time.sleep(2)

        print('setting pin 13 low')
        GPIO.output(channel, GPIO.LOW)
        time.sleep(2)
        

> print()

but some pins not working,  you can see that my code is just set a pin up and down,
the pin12 35 38 40 works fine, but the others maintain 1.6V.    

why is that??

the pin configuration is shown below
[image]

swarren, 2020-09-21:
I've personally tested all the pins on NX with a logic analyzer, so I know they do all work. Are you sure you're counting the pins correctly, i.e. using the correct physical pins? This has been the solution every time a similar issue has been reported in the past. BTW, for Jetson.GPIO to work (for plain GPIO usage, i.e. aside from the hardware PWM feature), you don't need to set any configuration via Jetson-IO. Did you ever change anything in Jetson-IO and save the result? If so, perhaps try a fresh SD card and see if that solves anything.

swarren, 2020-09-21:
BTW, which exact hardware are you using; I assume the Jetson NX Developer Kit, which uses an SD card for storage, and in particular you aren't using:
- Jetson NX module with built-in eMMC storage.
- Jetson NX module (either SD or eMMC SKU) on Jetson Nano baseboard.

I have only tested the Jetson NX Developer Kit (using SD storage). The other combinations should work identically, but I haven't verified them myself.

guizilaile23, 2020-09-23:
@swarren 
Thanks bro, i have change the Jetson-IO to enable SPI, and change it back, i don't know whether it will influence the pins,  by the way,  my device is Jetson NX Developer Kit (using SD storage), sorry to forgot the whole information,   i will re-flash the SD card and try again.

swarren, 2020-09-23:
Ah, pin 13 is an SPI pin, so that might be the issue. I will try and test for any interaction between Jetson.GPIO and Jetson-IO on that pin.

swarren, 2020-09-23:
I don't currently have access to a system with pin 13 connected to anything, but am trying to get that set up.

However, I did the same thing with pins 31 and 32; I used Jetson-IO to configure their alternate function, and Jetson.GPIO was able to use them as GPIOs without any issue as expected. It might be that the configuration for pin 13 has something special about it; we'll see if/when I can get the hardware configured.

swarren, 2020-10-06:
I have attempted to reproduce this problem on a Jetson NX, and everything appears to be working fine. Are you sure you haven't damaged your Jetson by e.g. connecting a GPIO pin to a 5V rather than 3.3V signal, or connecting two output pins together? What are you using to test the pin output; LED, multimeter, oscilloscope, some other hardware?

I tried the following, for each of pins 12 and 13:
- run_sample.sh simple_output.py
- Run Jetson-IO to reconfigure pin 13 to enable SPI, without rebooting
- run_sample.sh simple_output.py
- Reboot
- run_sample.sh simple_output.py
- Run Jetson-IO to reconfigure pin 13 to disable SPI, without rebooting
- run_sample.sh simple_output.py
- Reboot
- run_sample.sh simple_output.py

swarren, 2020-10-13:
I plan to close this bug in the next few days unless more information appears, since I can't reproduce it. Sorry.

swarren, 2020-10-23:
Closing stale bug.

closed 2020-10-23 by swarren

guizilaile23, 2021-01-14:
@swarren  sorry to bother you again.  after reflash the system,  i use a oscilloscope to capture the PWM, runing the "simple_out.py" in GPIO samples, which is neayly same with the code i wrote at the top. and i got figures.

for pin 12 35 38 40, works well.
PWM 
[image]

but for others, i got this.
[image]

 it seems that i need a pull-up resistance.

swarren, 2021-01-14:
(I'm a bit confused that you mention PWM, since most of the pins you mention don't support it. I assume you're just using plain GPIO output since you mention simple_out.py.)

I was able to reproduce an issue like this on pin 12, using simple_out.py; the GPIO output would toggle between 0 and 1.6V rather than 0 and 3.3V. However, I have only been able to capture it using a multimeter so far, so I can't view the full waveform to see if it matches yours.

I believe the issue is a hardware issue caused by static buildup/discharge. Often, the issue would start when I touched the wire attached to the GPIO pin, sometimes with a small spark from static. The issue would usually resolve itself after a few seconds, without any software change at all, so I know this is purely a hardware issue.

swarren, 2021-01-14:
Another thing I found was that it was much easier to reproduce the issue with *just* my multimeter attached to the GPIO output, but as soon as I attached my oscilloscope, the problem wouldn't ever reproduce. I wonder if my meter has very different impedance than the scope? If so, perhaps connecting e.g. a 10K resistor from the GPIO output to ground would solve the problem; perhaps the GPIO level shifters on the Jetson NX need some amount of current to flow in order to operate well?

(Be careful when trying this. Don't use a resistor with too low of a value; that would cause too much current to flow out of Jetson's GPIO pin. A 10k resistor should be fine, since that would yield just 0.33mA, which I believe should be fine.)

swarren, 2021-01-15:
Perhaps this is related to the behaviour of the level translators in Jetson? See the following application note:
https://developer.nvidia.com/embedded/downloads
Search for document "Jetson Nano Developer Kit 40-Pin Expansion Header GPIO Usage Considerations Applications Note".

guizilaile23, 2021-01-20:
@swarren  Thanks very much,  i will try it later~~~~

swarren, 2021-01-20:
As mentioned, I can reproduce this issue (or something similar) while measuring voltages using my multi-meter. One of our hardware designers suggested placing a 33k resistor in series with the probe on the meter, to reduce the load that the meter has on Jetson. I tried this, and that solved the issue for me. (I actually tried 22k and 47k resistors since I didn't have 33k).

Perhaps if you place a resistor between the Jetson GPIO pin and your scope's positive probe it will solve your problem too? If so, this is evidence that the issue is explained by the document I linked in the previous comment .