NVIDIA/jetson-gpio #45: Jetson.GPIO on Connect Tech astro carrier

FSUSE, 2020-10-01:
Hello, 

I am currently using this library. I am using the Astro carrier board from connect Tech. Is there a way to run this library on the Astro?

I get the following message:
[image]

Thank you

swarren, 2020-10-01:
For third-party boards, it's up to the vendor to implement support in Jetson.GPIO if they wish.

However, it doesn't look like there's any 40-pin GPIO expansion connector on the Astro board, so it may not make sense to support this board. The intent of Jetson.GPIO is to provide compatibility for software written to access a Raspberry Pi's 40-pin GPIO expansion connector. At the very least, if support for Astro was added to Jetson.GPIO, the pin numbering wouldn't be compatible with any existing software.

Can you describe your use-case a bit, e.g.:
* Are you writing new software or are you porting something existing?
* Do you care about compatibility of pin numbers with existing software or other board, or not?
* Do you explicitly want to use the Jetson.GPIO API, or did you try it simply because it was available. Perhaps an alternative API or mechanism would be suitable if it existed?

Thanks.

FSUSE, 2020-10-01:
Hi swarren, 

thank you for your reply
I want to write a software to control the GPIOs of Astro or Jetson. 
I can control the GPIO pins via the terminal. 

The program is implemented in an image processing software. 

Therefore I wanted to use this library.

swarren, 2020-10-02:
OK, you'll need to talk to Connect Tech re: having them add support to the library for the Astro board.

I guess they can invent a different pin naming scheme for this board, since it doesn't have a 40-pin header, and I guess leave some of the pin lookup table columns blank (e.g. the BCM mode values won't make any sense).

closed 2020-10-02 by swarren