NVIDIA/jetson-gpio #46: cp: cannot stat 'lib/python/Jetson/GPIO/99-gpio.rules': No such file or directory

hnwangkg-ezio, 2020-10-29:
cp: cannot stat 'lib/python/Jetson/GPIO/99-gpio.rules': No such file or directory

hnwangkg-ezio, 2020-10-29:
not /lib/python this directory

swarren, 2020-10-29:
When does this problem occur; what command did you run when you saw the problem?

swarren, 2020-10-29:
Perhaps you're running the commands in the README.md file? If so, make sure you run all the commands after using `cd` to go to the directory containing the Jetson.GPIO source; the same one the README file is in.

hnwangkg-ezio, 2020-10-29:
yes, i running the commands.and 
sudo python3 setup.py install
sudo groupadd -f -r gpio
sudo usermod -a -G gpio your_user_name

but not lib/python/ this dir..

swarren, 2020-10-29:
lib/python/ definitely does exist in the source (https://github.com/NVIDIA/jetson-gpio/tree/master/lib/python). It sounds like your git clone is corrupted; perhaps you can delete it and download the code again?

hnwangkg-ezio, 2020-10-29:
I downloaded it again, but the /lib/python directory is still not generated

hnwangkg-ezio, 2020-10-29:
It was my mistake, I regarded lib/python as /lib/python

hnwangkg-ezio, 2020-10-29:
WARNING: Carrier board is not from a Jetson Developer Kit.
WARNNIG: Jetson.GPIO library has not been verified with this carrier board,
WARNING: and in fact is unlikely to work correctly.
Is this warning normal?

swarren, 2020-10-29:
Which Jetson board do you have (which exact module and version, and which exact baseboard and version)? Which version of Jetson.GPIO are you using?

hnwangkg-ezio, 2020-10-29:
Jetson Nano 4GB Developer Kit, and my img is jetson-nano-4gb-jp441-sd-card-image
i think my Jetson.GPIO version is :
Installed /usr/local/lib/python3.6/dist-packages/Jetson.GPIO-2.0.12-py3.6.egg
Processing dependencies for Jetson.GPIO==2.0.12
Finished processing dependencies for Jetson.GPIO==2.0.12

swarren, 2020-10-29:
Please run the following commands and paste the results. Thanks.

ls -l /proc/device-tree/chosen/plugin-manager/ids

cat /proc/device-tree/compatible

python3
import Jetson.GPIO
print(Jetson.GPIO.VERSION)
print(Jetson.GPIO.__file__)

hnwangkg-ezio, 2020-10-29:
root@wangkg-desktop:/home/wangkg/wangkg/2-ch_Realy_for_JetsonNano# ls -l /proc/device-tree/chosen/plugin-manager/ids
total 0
-r--r--r-- 1 root root 26 10月 30 00:42 3448-0000-200
drwxr-xr-x 3 root root  0 10月 30 00:42 connection
-r--r--r-- 1 root root  4 10月 30 00:42 name
root@wangkg-desktop:/home/wangkg/wangkg/2-ch_Realy_for_JetsonNano# cat /proc/device-tree/compatible
nvidia,p3449-0000-b00+p3448-0000-b00nvidia,jetson-nanonvidia,tegra210root@wangkg-desktop:/home/wangkg/wangkg/2-ch_Realy_for_JetsonNano# 
root@wangkg-desktop:/home/wangkg/wangkg/2-ch_Realy_for_JetsonNano# 
root@wangkg-desktop:/home/wangkg/wangkg/2-ch_Realy_for_JetsonNano# python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import Jetson.GPIO
WARNING: Carrier board is not from a Jetson Developer Kit.
WARNNIG: Jetson.GPIO library has not been verified with this carrier board,
WARNING: and in fact is unlikely to work correctly.
>>> print(Jetson.GPIO.VERSION)
2.0.11
>>> print(Jetson.GPIO.file)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'Jetson.GPIO' has no attribute 'file'
>>>

swarren, 2020-10-29:
Are you using the baseboard that came with the Jetson module, or a 3rd-party baseboard?

If using the Jetson developer kit baseboard: There appears to be some problem detecting the baseboard. This may be a software or hardware problem. Either way, this problem is unrelated to Jetson.GPIO itself. You'll need to use the NVIDIA forums to get support to resolve this problem. See https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/70

If using a 3rd-party baseboard: Jetson.GPIO does not currently support any 3rd-party baseboards. Contact your baseboard vendor to get them to add support to Jetson.GPIO for their baseboard.

BTW, Python picked up Jetson.GPIO 2.0.11, rather than 2.0.12. This probably means that you haven't correctly set PYTHONPATH to pick up the version you installed, but instead are using the older version that is part of L4T. This issue isn't the cause of the problem you're having, but I mention is as an "FYI".

Closing this bug since the underlying issue is outside of Jetson.GPIO.

closed 2020-10-29 by swarren

hnwangkg-ezio, 2020-10-29:
using the baseboard that came with the Jetson module， tks.