NVIDIA/jetson-gpio #49: Update for Jetson Xavier NX

aldrichg9, 2021-01-29:
I wonder if there is an updated for the Xavier NX?

e.g. I want to use pin 445, but it is not listed in gpio_pin_data.py

swarren, 2021-01-29:
Jetson.GPIO already fully supports Jetson Xavier NX. There's no update available beyond what's already present in this repository or on PyPI.

Quoting from the original bug description before the edit:
> e.g. I want to use pin 445, but it is not listed in gpio_pin_data.py

Note that Jetson.GPIO deliberately only supports pins attached to the 40-pin GPIO expansion connector. I'm not sure what "pin 445" refers to; which device or connector is that pin part of? Thanks.

aldrichg9, 2021-01-29:
[image]

swarren, 2021-01-29:
That is Linux GPIO number 445. In turn, that's the following GPIO number for Jetson.GPIO:
* BOARD mode: 12 (from the pin number on the connector/header).
* BCM mode: 18.

aldrichg9, 2021-01-29:
> That is Linux GPIO number 445. In turn, that's the following GPIO number for Jetson.GPIO:
> 
> * BOARD mode: 12 (from the pin number on the connector/header).
> * BCM mode: 18.

Thanks swarren for your replies.

swarren, 2021-02-04:
@aldrichg9 I believe we can close this bug now? Thanks.

aldrichg9, 2021-02-04:
> @aldrichg9 I believe we can close this bug now? Thanks.

Of course!  Thank you for your help.

closed 2021-02-04 by swarren