NVIDIA/jetson-gpio #50: AGX WARNING: Carrier board is not from a Jetson Developer Kit

razvanphp, 2021-02-25:
This is actually identical to #32, but only happening on Jetson AGX Xavier with 32 GB eMMC.

```
jetson@agx:~$ sudo python3
Python 3.8.5 (default, Jul 28 2020, 12:59:40)
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import Jetson.GPIO as GPIO
WARNING: Carrier board is not from a Jetson Developer Kit.
WARNNIG: Jetson.GPIO library has not been verified with this carrier board,
WARNING: and in fact is unlikely to work correctly.
>>> GPIO.JETSON_INFO
{'P1_REVISION': 1, 'RAM': '16384M', 'REVISION': 'Unknown', 'TYPE': 'Jetson Xavier', 'MANUFACTURER': 'NVIDIA', 'PROCESSOR': 'ARM Carmel'}
>>> GPIO.VERSION
'2.0.16'
>>>
jetson@agx:~$ cat /etc/nv_tegra_release
# R32 (release), REVISION: 5.0, GCID: 25531747, BOARD: t186ref, EABI: aarch64, DATE: Fri Jan 15 23:21:05 UTC 2021
jetson@agx:~$
jetson@agx:~$ cat /proc/device-tree/compatible
nvidia,galennvidia,jetson-xaviernvidia,p2822-0000+p2888-0001nvidia,tegra194
jetson@agx:~$ ls -l /proc/device-tree/chosen/plugin-manager/ids
total 0
-r--r--r-- 1 root root 25 Feb 25 21:40 2888-0004-400
-r--r--r-- 1 root root 25 Feb 25 21:40 framos-imx464-0
-r--r--r-- 1 root root  4 Feb 25 21:40 name
jetson@agx:~$
```

The strange thing, this particular board worked before, so we don't really understand what happened and what can be the problem/debug actions. HDMI for example is working, so the board is clearly connected.

Thank you in advance!
R

anhmiuhv, 2021-02-25:
my directory looks like this on JAX:
```
2822-0000-700  2888-0004-400  name                                                                                                                                              
ubuntu@tegra-ubuntu:~$ 
```
Could you have plugged the module into a different carrier board?

razvanphp, 2021-02-25:
No, don’t think another carrier board for AGX even exists, it’s the developer kit.

Is there any way to further debug this?

We will try a full jetpack reinstall tho...

Thank you!
R

anhmiuhv, 2021-02-26:
If you still have problems, pls comment so I can ask someone on our team. Do you have anything plugged into the devkit?

razvanphp, 2021-02-26:
we only had a framos camera board and the wifi m2 card, but we tried to remove them both and the same thing.... really strange, we get 5V on the GPIO relevant pins.

anhmiuhv, 2021-02-26:
Do you mean that this happens? https://github.com/NVIDIA/jetson-gpio/issues/32#issuecomment-602069981

razvanphp, 2021-02-26:
no, I mean that we still get 5V on the pins when Jetson is powered on, so we can only assume that things still work, but they are not detected right.

anhmiuhv, 2021-02-26:
Yes, then definitely post on the forum. Still should try to test out some GPIO pin using Jetson.GPIO to see if it works.

razvanphp, 2021-03-01:
any idea how can I manually provide the dev board id to Jetson.GPIO? basically avoiding automatic detection now.

anhmiuhv, 2021-03-01:
you can modify this line. https://github.com/NVIDIA/jetson-gpio/blob/6f8250ae6713b5ccf439840b56c15b42bd460a6a/lib/python/Jetson/GPIO/gpio.py#L70

Also I believe this is just a warning. So. i think it still works as-it

razvanphp, 2021-03-02:
Thanks, I added a print after that line and it seems to be detecting:

```
model JETSON_XAVIER
jetson_info {'P1_REVISION': 1, 'RAM': '16384M', 'REVISION': 'Unknown', 'TYPE': 'Jetson Xavier', 'MANUFACTURER': 'NVIDIA', 'PROCESSOR': 'ARM Carmel'}
_channel_data_by_mode {'BOARD': {7: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f86444b38>, 11: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f86444b70>, 12: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f86444c88>, 13: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f86444cf8>, 15: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f86444d68>, 16: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f86444dd8>, 18: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f86444e48>, 19: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f86444eb8>, 21: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f86444f28>, 22: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f86444f98>, 23: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd048>, 24: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd0b8>, 26: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd128>, 29: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd198>, 31: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd208>, 32: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd278>, 33: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd2e8>, 35: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd358>, 36: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd3c8>, 37: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd438>, 38: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd4a8>, 40: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd518>}, 'BCM': {4: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd588>, 17: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd5f8>, 18: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd668>, 27: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd6d8>, 22: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd748>, 23: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd7b8>, 24: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd828>, 10: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd898>, 9: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd908>, 25: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd978>, 11: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dd9e8>, 8: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dda58>, 7: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863ddac8>, 5: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863ddb38>, 6: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863ddba8>, 12: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863ddc18>, 13: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863ddc88>, 19: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863ddcf8>, 16: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863ddd68>, 26: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dddd8>, 20: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863dde48>, 21: <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863ddeb8>}, 'CVM': {'MCLK05': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863ddf28>, 'UART1_RTS': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863ddf98>, 'I2S2_CLK': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0048>, 'PWM01': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e00b8>, 'GPIO27': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0128>, 'GPIO8': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0198>, 'GPIO35': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0208>, 'SPI1_MOSI': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0278>, 'SPI1_MISO': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e02e8>, 'GPIO17': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0358>, 'SPI1_CLK': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e03c8>, 'SPI1_CS0_N': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0438>, 'SPI1_CS1_N': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e04a8>, 'CAN0_DIN': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0518>, 'CAN0_DOUT': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0588>, 'GPIO9': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e05f8>, 'CAN1_DOUT': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0668>, 'I2S2_FS': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e06d8>, 'UART1_CTS': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0748>, 'CAN1_DIN': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e07b8>, 'I2S2_DIN': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0828>, 'I2S2_DOUT': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0898>}, 'TEGRA_SOC': {'SOC_GPIO42': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0908>, 'UART1_RTS': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0978>, 'DAP2_SCLK': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e09e8>, 'SOC_GPIO44': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0a58>, 'SOC_GPIO54': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0ac8>, 'CAN1_STB': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0b38>, 'SOC_GPIO12': <Jetson.GPIO.gpio_pin_data.ChannelInfo object at 0x7f863e0ba8> [... cut ...]
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --
```
so actually our `adafruit_pca9685` I2C device is detected, but there seems to be a problem in python...

Can you tell us what the output of `ls /proc/device-tree/chosen/plugin-manager/ids` should be for Jetson AGX Dev Board 32GB? or maybe the warning is "normal"?

anhmiuhv, 2021-03-02:
This is what it should look like. https://github.com/NVIDIA/jetson-gpio/issues/50#issuecomment-786251085

Jetson_xavier is the right device. Clara is another product based on jetson xavier.

razvanphp, 2021-03-02:
ok, so then it should work... since the detection is fine.

do you know another "manual" way to test the I2C through Jetson.GPIO? using the adafruit SDK we get `OSError: [Errno 121] Remote I/O error` from the linked-above github issue.... and again, this worked before: we even tried another adafruit device (not working on AGX) or this device on another jetson nano (works).

We're running out of ideas... :(

anhmiuhv, 2021-03-02:
Jetson GPIO does not support I2C sorry

anhmiuhv, 2021-03-02:
Please post your question on the forum https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/70 so our engineer can take a further look. Jetson GPIO is only responsible for GPIO functionalities and PWM on the 40-pin headers.

razvanphp, 2021-03-04:
Thank you, we did open another thread on the forum, we found out we override some EEPROM values while installing the Framos cameras.

Would you be so kind to show us how those registers should look like on AGX?

```
sudo i2cdump -y 0 0x56
sudo i2cdump -y 0 0x55
```

Also, we are actually using the I2C board to control PWM motors, you say it's possible to drive a servo from the GPIO PWM, or is it only for LEDs?

anhmiuhv, 2021-03-04:
sudo i2cdump -y 0 0x56

No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 01 00 ff 00 06 0b 00 00 07 4a 00 00 00 00 00 00    ?...??..?J......
10: 00 01 1c 14 36 39 39 2d 38 32 38 32 32 2d 30 30    .???699-82822-00
20: 30 30 2d 37 30 30 20 4a 2e 30 00 00 00 00 00 00    00-700 J.0......
30: 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
40: ff ff ff ff ff ff ff ff ff ff 31 34 32 31 34 32    ..........142142
50: 30 30 30 35 31 34 38 00 00 00 00 00 00 00 00 00    0005148.........
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
90: 00 00 00 00 00 00 46 46 46 46 ff ff 46 46 ff ff    ......FFFF..FF..
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
b0: ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 e6    ...............?

anhmiuhv, 2021-03-04:
Their is nothing on 0x55

anhmiuhv, 2021-03-04:
I think it is possible to control from jax. you can try

anhmiuhv, 2021-03-04:
```
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 01 00 ff 00 06 0b 00 00 07 4a 00 00 00 00 00 00    ?...??..?J......
10: 00 01 1c 14 36 39 39 2d 38 32 38 32 32 2d 30 30    .???699-82822-00
20: 30 30 2d 37 30 30 20 4a 2e 30 00 00 00 00 00 00    00-700 J.0......
30: 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
40: ff ff ff ff ff ff ff ff ff ff 31 34 32 31 34 32    ..........142142
50: 30 30 30 35 31 34 38 00 00 00 00 00 00 00 00 00    0005148.........
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
90: 00 00 00 00 00 00 46 46 46 46 ff ff 46 46 ff ff    ......FFFF..FF..
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
b0: ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 e6    ...............?
```

razvanphp, 2021-03-05:
ok, we fixed (rebuilt) the EEPROM and now the warning is gone, board is detected and we have all files in here:
```
jetson@agx:~$ ls -al /proc/device-tree/chosen/plugin-manager/ids
total 0
drwxr-xr-x 2 root root  0 Mar  4 13:05 .
drwxr-xr-x 7 root root  0 Mar  4 13:05 ..
-r--r--r-- 1 root root 25 Mar  4 13:05 2822-0000-700
-r--r--r-- 1 root root 25 Mar  4 13:05 2888-0004-400
-r--r--r-- 1 root root 25 Mar  4 13:05 framos-adapter
-r--r--r-- 1 root root  4 Mar  4 13:05 name
jetson@agx:~$
```
But, unfortunately the PWM does not give any signal on pins 15 & 18, even after we enabled it on the `getson-io.py` script, we checked on the oscilloscope. Do we need to enable it on the DTB? Or how to manually link the pwm generator to the pins?

anhmiuhv, 2021-03-05:
You can use Jetson-IO tool

anhmiuhv, 2021-03-05:
Did you restart?

anhmiuhv, 2021-03-05:
```
•Save and reboot to reconfigure pins: Creates a new DTB by applying a Device Tree overlay for the configuration. It then updates the configuration file for booting Linux (/boot/extlinux/extlinux.conf) and reboots the developer kit.
Note that after configuration file has been updated you can still boot the developer kit to the previous configuration, because Jetson‑IO updates extlinux.conf by adding a new entry for the new configuration. The previous configuration is still in the file. When the target is booted, the bootloader lets you select either configuration (or any other that is defined in the file).
```

Did you boot into the right extlinux entry?

anhmiuhv, 2021-03-07:
you can enable pwm manually using this line. https://github.com/NVIDIA/jetson-gpio/blob/532bd38c5becc44cfcebf8ec3ee3c8e87818dbad/samples/test_all_apis.py#L33

the python file also show how to use pwm.

razvanphp, 2021-03-08:
Yes, we did restart as it is suggested by the python script. We also run the busybox command manually, still flat line on the oscilloscope. Strange thing is, the 5V and 3.3V pins are fine, it just seems the other GPIO pins (I2C and PWM) are not working anymore, even tho the board is detected.

Is there any way to debug this further, even if there is a hardware problem? Maybe even sending the board back to nvidia for repairs? Not sure how to prove the warranty in this case to the reseller tho.

[image]

anhmiuhv, 2021-03-08:
For hardware debug please post your question on the forum. If you believe your product require warranty, please go to this website, https://www.nvidia.com/en-us/support/warranty/

anhmiuhv, 2021-03-08:
Also did you test without the framos board? Do the gpio functionality still works normally? What script did you run to test pwm?

razvanphp, 2021-03-08:
We now tried with your test script, here it fails on setting one pin to high and then when we left in only PWM tests, it also fails:

```
jetson@agx:~$ sudo python3 test.py
Testing test_warnings_off
Testing test_warnings_on
Testing test_setup_one_board
Testing test_setup_one_bcm
Testing test_setup_one_cvm
Testing test_setup_one_tegra_soc
Testing test_setup_one_out_no_init
Testing test_setup_one_out_high
Testing test_setup_one_out_low
Testing test_setup_many_out_no_init
Testing test_setup_many_out_one_init
Testing test_setup_many_out_many_init
Testing test_setup_one_in
Testing test_setup_one_in_pull
/home/jetson/.local/lib/python3.6/site-packages/Jetson/GPIO/gpio.py:370: UserWarning: Jetson.GPIO ignores setup()'s pull_up_down parameter
  warnings.warn("Jetson.GPIO ignores setup()'s pull_up_down parameter")
Testing test_setup_many_in
Testing test_setup_all
Testing test_cleanup_one
Testing test_cleanup_many
Testing test_cleanup_all
Testing test_input
Testing test_output_one
Testing test_output_many_one_value
Testing test_output_many_many_value
Testing test_out_in_init_high
Traceback (most recent call last):
  File "test.py", line 717, in <module>
    test()
  File "test.py", line 381, in test_out_in_init_high
    assert(val == GPIO.HIGH)
AssertionError
jetson@agx:~$ vi test.py
jetson@agx:~$ sudo python3 test.py
Testing test_pwm_multi_duty
Traceback (most recent call last):
  File "test.py", line 250, in <module>
    test()
  File "test.py", line 175, in test_pwm_multi_duty
    assert min_ct <= count <= max_ct
AssertionError
jetson@agx:~$
```

anhmiuhv, 2021-03-08:
Did you try this script https://github.com/NVIDIA/jetson-gpio/blob/master/samples/simple_pwm.py ?

razvanphp, 2021-03-08:
We just tried, it says PWM running but nothing on the osciloscope.

On the jetson NX on another devboard, the test script works:

[image]

anhmiuhv, 2021-03-08:
Then I think it is hardware problem then. I am not familiar with hardware debug so you will need to ask on the forum. Also try to access the link for the warranty for more information.

closed 2021-03-08 by anhmiuhv