NVIDIA/jetson-gpio #52: Containerized gpio access

sashabel, 2021-04-26:
Hello, I have a question. I'm interested in operating GPIO on Jetson NANO from within container and then from K3s.
I encountered errors when tried to activate `simple_out.py` example from docker container.

My dockerfile is:
```
#FROM ubuntu:20.04
#FROM python:2
FROM nvcr.io/nvidia/l4t-base:r32.4.3

ARG uid
ARG gid
ARG gid_gpio
RUN groupadd -f -r -g $gid_gpio gpio
RUN groupadd -f -r -g 1000 user

RUN apt-get update && \
    apt-get install -y python-pip && \
    pip install Jetson.GPIO

# gpio needs to be main group, otherwise we get permission problems
RUN useradd -M --uid $uid -g $gid_gpio user --groups uucp,$gid,user 
RUN mkdir -p /etc/sudoers.d \ 
   && echo 'user ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers.d/user \
   && echo 'Defaults exempt_group+=user' >> /etc/sudoers.d/user \
   && chmod a=r,o= /etc/sudoers.d/user

WORKDIR /app
COPY simple_out.py /app

CMD ["/bin/sh", "-c", "python /app/simple_out.py"]
```

I am running the container with following script:
```
#!/bin/bash

source ../configure.sh
 
docker run --rm  -it \
   --runtime=nvidia --gpus all\
  -v /sys:/sys \
  --device /dev/spidev0.0:/dev/spidev0.0:rw \
  --group-add $(cut -d: -f3 < <(getent group gpio)) "poc/gpio:${POC_VER}_${MY_KUBECTL_VERSION}_${MY_TARGETPLATFORM}" /bin/bash
 ```

and the errors are printed:
```
root@46ede1d99ba7:/app# python simple_out.py 
Traceback (most recent call last):
  File "simple_out.py", line 22, in <module>
    import RPi.GPIO as GPIO
  File "/usr/local/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/usr/local/lib/python2.7/dist-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/usr/local/lib/python2.7/dist-packages/Jetson/GPIO/gpio.py", line 70, in <module>
    model, JETSON_INFO, _channel_data_by_mode = gpio_pin_data.get_data()
  File "/usr/local/lib/python2.7/dist-packages/Jetson/GPIO/gpio_pin_data.py", line 373, in get_data
    with open(compatible_path, 'r') as f:
IOError: [Errno 2] No such file or directory: '/proc/device-tree/compatible'
```

I have tried with the NVIDIA base container (`nvcr.io/nvidia/l4t-base:r32.4.3`) but it didn't help. I understand that the `/sys` contains files that should be mapped into container, (`/proc/device-tree/compatible`  is a symlink to one of the files in that directory)

Any ideas ?

anhmiuhv, 2021-04-26:
So if you map proc also, does it work?

sashabel, 2021-04-26:
No,

tried 
```
 docker run --rm  -it \
   --runtime=nvidia --gpus all\
  -v /sys:/sys \
  -v /proc/device-tree/compatible:/proc/device-tree/compatible \
  --device /dev/spidev0.0:/dev/spidev0.0:rw \
  --group-add $(cut -d: -f3 < <(getent group gpio)) "poc/gpio:${POC_VER}_${MY_KUBECTL_VERSION}_${MY_TARGETPLATFORM}" 
  
```

but still

```
Traceback (most recent call last):
  File "/app/simple_out.py", line 22, in <module>
    import RPi.GPIO as GPIO
  File "/usr/local/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/usr/local/lib/python2.7/dist-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/usr/local/lib/python2.7/dist-packages/Jetson/GPIO/gpio.py", line 70, in <module>
    model, JETSON_INFO, _channel_data_by_mode = gpio_pin_data.get_data()
  File "/usr/local/lib/python2.7/dist-packages/Jetson/GPIO/gpio_pin_data.py", line 373, in get_data
    with open(compatible_path, 'r') as f:
IOError: [Errno 2] No such file or directory: '/proc/device-tree/compatible'
```

anhmiuhv, 2021-04-26:
This command run on Jetson Nano?

sashabel, 2021-04-26:
Yes

anhmiuhv, 2021-04-26:
Can you find the file manually inside docker? I am not quite familiar with docker. 

Maybe you can modify the code here https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py#L369

so that it return JETSON_NANO data directly

anhmiuhv, 2021-04-26:
Maybe you can map /proc:/proc and run container in privilleged mode

sashabel, 2021-04-26:
I ran the `gpio_pin_data.py` from within container, there was no output nor errors printed
```
root@f36ce0f29510:/app# python gpio_pin_data.py
root@f36ce0f29510:/app# python simple_out.py
Traceback (most recent call last):
  File "simple_out.py", line 22, in <module>
    import RPi.GPIO as GPIO
  File "/usr/local/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/usr/local/lib/python2.7/dist-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/usr/local/lib/python2.7/dist-packages/Jetson/GPIO/gpio.py", line 70, in <module>
    model, JETSON_INFO, _channel_data_by_mode = gpio_pin_data.get_data()
  File "/usr/local/lib/python2.7/dist-packages/Jetson/GPIO/gpio_pin_data.py", line 373, in get_data
    with open(compatible_path, 'r') as f:
IOError: [Errno 2] No such file or directory: '/proc/device-tree/compatible'

```

anhmiuhv, 2021-04-26:
You have to change the code of the library. Set compatibles directly to jetson nano instead of having to read form file
```
compatibles='nvidia,jetson-nano'
```

anhmiuhv, 2021-04-26:
> Maybe you can map /proc:/proc and run container in privilleged mode

Also did you try this?

sashabel, 2021-04-26:
The privileged mode works fine, but it is not secured so if there is a way to run it in regular mode this is preferable.
Regarding `compatibles='nvidia,jetson-nano'` what file should I change ?

anhmiuhv, 2021-04-26:
This file. https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py#L369

You will probably need to find where the library is installed and change this python file Jetson/GPIO/gpio_pin_data.py.

sashabel, 2021-04-26:
I've overwrite the file as you have recommended:
```
COPY gpio_pin_data.py /usr/lib/python2.7/dist-packages/Jetson/GPIO/gpio_pin_data.py
COPY gpio_pin_data.py /usr/lib/python3/dist-packages/Jetson/GPIO/gpio_pin_data.py
```

but still `simple_out.py` works only in privileged mode

anhmiuhv, 2021-04-26:
what error does it say?

sashabel, 2021-04-26:
On my device:
```
$ cat /proc/device-tree/compatible ; echo
nvidia,p3449-0000-b00+p3448-0000-b00nvidia,jetson-nanonvidia,tegra210
```

After changing the file lines to be:
```
 # original:
 #   with open(compatible_path, 'r') as f:
 #       compatibles = f.read().split('\x00')
 # I changed to :
    compatibles='nvidia,p3449-0000-b00+p3448-0000-b00nvidia,jetson-nanonvidia,tegra210'
```

The error :
```
 File "/usr/local/lib/python2.7/dist-packages/Jetson/GPIO/gpio_pin_data.py", line 429, in get_data
    raise Exception('Could not determine Jetson Nano module revision')
Exception: Could not determine Jetson Nano module revision
```

anhmiuhv, 2021-04-26:
I guess you also need to overwrite line
https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py#L429

You can find the value in here.

'/proc/device-tree/chosen/plugin-manager/ids'

There should be some files like these (This is on my NX so on yours are different). Just try to overwrite revision with one of them to see which one works.
```
3509-0000-100  3668-0000-200  name
```

sashabel, 2021-04-26:
Ok, I've changed
```      
        # was :
        # module_id = find_pmgr_board('3448')
        # now:
        module_id = '3448-0000-400'
```

And it works,
thanks

closed 2021-04-26 by sashabel

abuelgasimsaadeldin, 2021-07-22:
Hi @anhmiuhv @sashabel I am also trying to import RPi.GPIO on a docker container and the error I am facing is as follows:
```
FileNotFounError: [Errno 2] No such file or directory: '/proc/device-tree/compatible'
```

I am running the container using: 
```
'sudo' docker run .... -v /sys:/sys -v proc:/proc
```

And I have edited the module id similar to what @sashabel has done and I still receive this error. Please what other steps can I perform to access my Nano's GPIO pins and successfully import RPi.GPIO on a container environment?