NVIDIA/jetson-gpio #54: GPIO setup with initial state dips voltage down for ~300us

fabiomanz, 2021-05-14:
**Describe the bug**

There is a short voltage drop to 0V when a Pin is setup via the GPIO.setup function. This drop is around 300μs long and occures if you use the initial optional flag or if you put the pin HIGH afterwards. If you setup the pin multiple times in one script, preferably with a delay in between, you can also observe the behavior.

- Jetson.GPIO.VERSION: '2.0.16'
- L4T Version 32.5.1
- We are using a Jetson Xavier NX with our own carrier board.

**To Reproduce**
Steps to reproduce the behaviour:
1. Connect an oscilloscope to a GPIO pin of the Jetson
2. Setup the pin via this function: GPIO.setup(22, GPIO.OUT, initial=GPIO.HIGH)
3. Notice the behavior

**Expected behavior**
Because of the internal pull-up I think, the Jetson GPIO pin is at a steady 1.8V level. If you setup the pin as an output while setting initial to HIGH, then you should not see any difference in the pin level. But in reality, the level drops down to 0V which is not acceptable in our use case.

**Additional context**
Please look at the provided picture of the oscilloscope, you also need to know, that the output is level shifted to 5V.

[image]

anhmiuhv, 2021-05-14:
I think pull up is disabled when you switch to output. jetson-gpio basically set direction to out then call output(initial) if initial is set, that is why you see the drop.

Maybe you can try to use gpiod instead https://www.acmesystems.it/gpiod. To output high on pin 22 you should run

```
sudo gpioset gpiochip0 193=1
```

fabiomanz, 2021-06-03:
Hi, I didn't fix the issue, but found a workaround. Now I only initialise the pin when it is needed, which works for my application. The test with libgpiod showed that they don't have the issue. 
Thanks for your help,
Fabio

closed 2021-06-03 by fabiomanz