NVIDIA/jetson-gpio #56: Invalid Channels

kaloomte, 2021-06-13:
Hi, sorry for if i miss something.
This is my code for taking input from GPIO 10.

```
import Jetson.GPIO as GPIO 

class CheckGPIO():

    def __init__(self): 
        GPIO.setmode(GPIO.BOARD) 
        GPIO.setwarnings(False) 
        self.CONVEYOR_BACKWARD_PIN = 10 
        GPIO.setup(self.CONVEYOR_BACKWARD_PIN, GPIO.IN) 

    def _check_conveyor(self): 
        if GPIO.input(self.CONVEYOR_BACKWARD_PIN): 
            print("LOW") 
        else: 
            print("HIGH") 

checkclass = CheckGPIO() 
checkclass._check_conveyor()
```

But i got an error. Here is error log;

```
Traceback (most recent call last):
  File "karalama.py", line 55, in <module>
    check = CheckGPIO()
  File "karalama.py", line 22, in __init__
    GPIO.setup(self.CONVEYOR_BACKWARD_PIN, GPIO.IN)
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 358, in setup
    ch_infos = _channels_to_infos(channels, need_gpio=True)
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 121, in _channels_to_infos
    for c in _make_iterable(channels)]
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 121, in <listcomp>
    for c in _make_iterable(channels)]
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 104, in _channel_to_info_lookup
    raise ValueError("Channel %s is invalid" % str(channel))
ValueError: Channel 10 is invalid
```

anhmiuhv, 2021-06-13:
We dont support PIN 10. You can read this phthon file to see what pins are supported. https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py

It should be in the column where the list is “7,11,12,13,…”

kaloomte, 2021-06-16:
Thank you for answer.

closed 2021-06-16 by kaloomte