NVIDIA/jetson-gpio #58: Modified example `button_interrupt.py` makes my Jetson Xavier NX freeze and reboot

gustavovelascoh, 2021-07-21:
Hi,

running the example `button_interrupt.py` but changing the pin definitions to:

```
# Pin Definitions:
led_pin_1 = 12
led_pin_2 = 13
but_pin = 18
```

makes my Jetson Xavier NX freeze and reboot. No single print is shown.

Then I put a 1-second delay before the ```GPIO.add_event_detect(but_pin, GPIO.FALLING, callback=blink, bouncetime=10)``` line and the program reached the main loop. But after a handful of button pushings, it freezes again.

Here are my device details:

        ➜  ~ uname -a
        Linux xavier1-desktop 4.9.140-tegra #1 SMP PREEMPT Tue Oct 27 21:02:46 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux
        ➜  ~ head -n 1 /etc/nv_tegra_release
        # R32 (release), REVISION: 4.4, GCID: 23942405, BOARD: t186ref, EABI: aarch64, DATE: Fri Oct 16 19:37:08 UTC 2020
       ➜  ~ pip3 freeze | grep GPIO        
       Jetson.GPIO==2.0.11

However, I've just noticed the example uses ```import RPi.GPIO as GPIO``` instead of ```import Jetson.GPIO as GPIO```. Is it intended?

Thanks.

anhmiuhv, 2021-07-22:
Pin 1, 2 is not supported. You can see supported PIN here.
https://www.jetsonhacks.com/nvidia-jetson-xavier-nx-gpio-header-pinout/

RPi.GPIO is ok

gustavovelascoh, 2021-07-22:
Hi @anhmiuhv, thanks for your reply. I'm using pins 11, 12 and 13 as input, input, output respectively.

I asked about RPi vs Jetson because when printing `RPi.GPIO.VERSION` is different to what is printed by `Jetson.GPIO.VERSION`.

anhmiuhv, 2021-07-22:
So we do that so that when one import RPi.GPIO it also import Jetson.GPIO .

As for your other problems, can you try with the latest version of the code?

Does the code works? Or the code work but it after sometime it freeze?

anhmiuhv, 2021-07-22:
If you can get me the syslog after it reboot it would be great. You can get the previous boot log after it reboot using this command:
```
journalctl -b-1
```

gustavovelascoh, 2021-07-23:
> So we do that so that when one import RPi.GPIO it also import Jetson.GPIO .
> 
> As for your other problems, can you try with the latest version of the code?
> 
> Does the code works? Or the code work but it after sometime it freeze?

With the previous code, it never toggled the LED, then I put some prints and a delay and it started toggling the LED and even detected a couple of button presses but after 10 to 15 seconds it stops responding. Once the console was frozen but still was able to detect the falling edge from the button and toggle the second LED for about 5 seconds and then rebooted.

I'll try to test the latest code and enable the `journalctl` as it is not working, and let you know what I find

anhmiuhv, 2021-08-11:
Does it still happen?

anhmiuhv, 2021-12-08:
If you have solved it I will close this issue

closed 2021-12-08 by anhmiuhv