NVIDIA/jetson-gpio #59: PWM and interrupt cannot work together

zhouzhiwen2000, 2021-07-23:
Hi, I am using PWM to drive a beeper and 3 buttons as interrupts.
However, when the PWM is started, the interrupts no longer work.
Here is the code:
```

#!/usr/bin/env python
import rospy
import Jetson.GPIO as GPIO
import time
led_pin_1 = 19
led_pin_2 = 21

def key_process(channel):
    print("key pressed:")
    print(channel)

def listener():
    GPIO.setmode(GPIO.BOARD)
    GPIO.setwarnings(False)
    GPIO.setup([29,31,35], GPIO.IN) #buttons  
    GPIO.setup(led_pin_1, GPIO.OUT,initial=GPIO.HIGH)
    GPIO.add_event_detect(29, GPIO.RISING, callback=key_process, bouncetime=200)#key1
    GPIO.add_event_detect(31, GPIO.RISING, callback=key_process, bouncetime=200)#key2
    GPIO.add_event_detect(35, GPIO.RISING, callback=key_process, bouncetime=200)#key3
    GPIO.setup(33, GPIO.OUT,initial=GPIO.LOW) #BEEPER
    p = GPIO.PWM(33, 2500)
    p.start(50)
    try:
        while True:
            time.sleep(1)
    finally:
        p.stop()  # cleanup all GPIOs
        GPIO.cleanup()

if __name__ == '__main__':
    listener()

```

anhmiuhv, 2021-07-27:
are you using the latest version?

zhouzhiwen2000, 2021-07-31:
Yes,the laest.

anhmiuhv, 2021-08-09:
Also what device is this?

anhmiuhv, 2021-08-10:
because Jetson GPIO cannot control pull-up/down maybe that is why the button event is failing. Can you try connect the button with ground and then connect button with 3.3V to see if the event activate?

zhouzhiwen2000, 2021-08-11:
Xavier NX the button has external pull-up resistors.

zhouzhiwen2000, 2021-08-11:
Splitting it into two py files solves the problem.

anhmiuhv, 2021-08-11:
ok i guess i can close this