NVIDIA/jetson-gpio #60: GPIO interrput

shaonianlg, 2021-08-12:
I use gpio by Jetson GPIO Python library,the code is here

`import Jetson.GPIO as GPIO
import time

edge_pin=15
plus_pin=31

def wait_egde(c):
GPIO.output(plus_pin,GPIO.HIGH)
time.sleep(0.001)
GPIO.output(plus_pin,GPIO.LOW)

def gpio_init():
GPIO.setmode(GPIO.BOARD)
GPIO.setup(edge_pin,GPIO.IN)
GPIO.setup(plus_pin,GPIO.OUT,initial=GPIO.LOW)
GPIO.add_event_detect(edge_pin,GPIO.RISING,callback=wait_egde)

def main():
gpio_init()

   while True:
if name == ‘main’:
main()`

I want to detect plus rising edge by gpio 15 interrupt,the plus frequency is 80Hz,in interrupt callback funtion output a plus by gpio 31.It seems that the plus frequency of gpio 31 is also 80Hz,but it’s not,it’s a unregular plus.Is there any solution?

anhmiuhv, 2021-08-12:
Can you try to use https://ss64.com/bash/nice.html to run your program with highest  priority?

anhmiuhv, 2021-08-12:
It is python code so interrupt might not run with strict frequency. So you might need to write your program in C or find a workaround using pwm for example.

shaonianlg, 2021-08-12:
> It is python code so interrupt might not run with strict frequency. So you might need to write your program in C or find a workaround using pwm for example.

I didn't add bouncetime in GPIO.add_event_detect,could it be the problem?

anhmiuhv, 2021-08-12:
You could try to do that to see if it fixes

anhmiuhv, 2021-12-08:
If you have solved it I will close this issue

closed 2021-12-08 by anhmiuhv