NVIDIA/jetson-gpio #61: GPIO on Jetson Xavier

imihassan, 2021-09-09:
Hi, 

I followed all the procedures mentioned by you to run the GPIO pins but I am not able to run it. What can be the issue can you guide me? What am I missing or doing wrong? Any help will be appreciated.

The procedure I followed is as shown below: 

Configure the GPIO expansion header for PWM or any other PIN, To configure run the following command. 
`sudo /opt/nvidia/jetson-io/jetson-io. 
`and follow this [link ](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/hw_setup_jetson_io.html)
Install GPIO Jetson using command
`sudo pip3 install Jetson.GPIO
`
Copy new rules
`sudo cp lib/python/Jetson/GPIO/99-gpio.rules /etc/udev/rules.d/
`
Run the following python code to toggle the voltage value. Check with a multimeter with GND and PWM (or whatever enabled) PINs
```
import Jetson.GPIO as GPIO
import time
GPIO.setmode(GPIO.BOARD)
channel = 15

GPIO.setup(channel, GPIO.OUT)

while True:
   GPIO.output(channel, GPIO.HIGH)
   
   time.sleep(1)
   GPIO.output(channel, GPIO.LOW)

   time.sleep(1)

GPIO.cleanup()
```

enfild, 2021-11-03:
Hi, I've met this too. I noticed that this can only be found on XAVIER NX with jetpack 4.6. On old JetPack and jetson nano all work normal

anhmiuhv, 2021-11-03:
Can you post the log here?

sanketh1691, 2021-11-22:
Hello, 

Can anyone post as to how to configure the pinmux for the PWM signals to reach the pins? I have configured using the (sudo /opt/nvidia/jetson-io/jetson-io.) command but it is still not working. I think pinmux is the only way but I am not able to understand on how to do it properly.

anhmiuhv, 2021-11-22:
after configuring you will need to reboot for the change to take effects.

anhmiuhv, 2021-11-22:
Just for testing, you can use a busybox command to configure as in the test comment sample here for tx2 nx
https://github.com/NVIDIA/jetson-gpio/blob/master/samples/test_all_apis.py#L120

sanketh1691, 2021-11-23:
Hi @anhmiuhv, thank you for your response. But yeah, I have rebooted the Jetson AGX Xavier after manually configuring the PWM pins using (/opt/nvidia/jetson-io/jetson-io.), no change, I was trying to get a servo motor work with a connection of pin 18 (GPIO.BOARD mode), a connection to ground and also a connection to 3.3v. 

So, I am still trying to understand how to get the system pinmux configured. I understood the steps to follow but not sure on what to change in Jetson_AGX_Xavier_pinmux_configuration template.xlms for the PWM signals and even when I made some changes to it, I was not able to generate those device tree files by clicking the Generate DT button. 

Any help would be greatly appreciated.

anhmiuhv, 2021-11-23:
please look at this thread on the forum:
https://forums.developer.nvidia.com/t/can-the-nvidia-drive-agx-xavier-be-used-without-the-use-of-external-microcontrollers/65775

anhmiuhv, 2021-11-23:
/opt/nvidia/jetson-io/jetson-io. should change your /boot/extlinux/extlinux.conf to use a new dtb. You can check whether extlinux has been changed to use a new dtb. Or you can use an oscilloscope to test whether PWM work

sanketh1691, 2021-11-23:
How to check extlinux has been changed or not? Is it in Tegra_for_Linux folder? If yes then for some reason I was not able to find that folder as well in my Jetson, so I had to download that particular folder which matches with my Jetpack version 4.6

anhmiuhv, 2021-12-08:
The folder should be on the Jetson already

imihassan, 2022-01-27:
Hi, I reinstalled the JetPack and followed the steps mentioned in the repo and I checked with a multimeter but no success. 
I checked the power pins of GPIOs that give proper voltage but GPIOs are not working. What can be the issue? How can I check the issue?

anhmiuhv, 2022-01-27:
multimeter might not work.
This is due to the sensitivity of Jetson's level translator circuit to the impedance/capacitance of the connected device.

You will need an analyzer with high input impedance and low probe wire length and capacitance. I personally found that the following work:
ASIX SIGMA2 (https://www.asix.net/dbg_sigma.htm).
My ancient HP logic analyzer's (16500B) digital and analog input boards.
An LED, a series resistor of around 150-300 Ohms, and some wires that can be connected to the GPIO pins. Note that Jetson's GPIO drive strength is very low, so if you use this method, the LED will be very dim. Make sure you don't use a too-large series resistor or you won't be able to see it. Make sure the resistor isn't too small or you may draw too much current.
A simple logic analyser https://www.sparkfun.com/products/18627

imihassan, 2022-01-27:
Ok. Thanks I can check the UART and test it.

When I checked GPIO folder with command ls /sys/class/gpio

I found these directories

export  gpiochip240  gpiochip248  gpiochip288  unexport

In [this youtube](https://www.youtube.com/watch?v=2LctZ8n1Wl0) tutorial at video time 17:20 to blink LED he is using gpio393 directory inside the /sys/class/gpio/   

I do not have this directory. Is there any installation error or something. I installed jetpack 4.5.1. 

Jetson GPIO is already installed in it. If I want to install from pip or setup it says already installed. 

I may be doing some mistake here. Can you help me in that.

anhmiuhv, 2022-01-27:
That directory will appear when you echo the correct value in to "export"
I think you can try by using 
echo 393 > /sys/class/gpio/export

anhmiuhv, 2022-04-18:
If you have no other comments, I will close this

closed 2022-04-18 by anhmiuhv