NVIDIA/jetson-gpio #62: Jetson.GPIO warning

basfari, 2021-12-08:
Hi guys, first of all thank you very much for your efforts.. I know that Jetson xavier nx developer kit has SD slot and no eMMC; while Jetson xavier nx module has eMMC and no SD slot.
Recently I’ve purchased from a 3rd party Jetson xavier nx developer kit which has eMMC and no SD slot (it boots from the internal storage): so my jetson has a customized carrier board.
Now when I use Jetson.GPIO, I get the warning:
```
WARNING: Carrier board is not from a Jetson Development Kit.
WARNING: Jetson.GPIO library has not been verified with this carrier board
WARNING: and in fact is unlikely to work correctly.
```

The ids of the board is:
```
>> ls /proc/device-tree/chosen/plugin-manager/ids
3668-0001-300   name   XXXX-XXXX-XXX
```
The content of 3668-0001-300 is:
`i2c@3160000:module@0x50`
and of XXXX-XXXX-XXX is:
`i2c@3160000:module@0x57`

The content of the compatible of the device-tree reads as follows:
```
>> cat /proc/device-tree/compatible
nvidia,p3449-0000+p3668-0000nvidia,p3449-0000+p3668-0001nvidia,p3509-0000+p3668-0000nvidia,p3509-0000+p3668-0001nvidia,tegra194
```

I believe the ids of the board should have been:
```
>> ls /proc/device-tree/chosen/plugin-manager/ids
3668-0000-300   name   3509-0000-100
```

The Jetpack version is the latest: v4.6
and Jetson.GPIO version is also the latest.

What do you recommend me to do? Is there a workaround to solve this warning?
As the compatible file shows that it is compatible with jetson xavier nx developer kit. Can I ignore the warning?

Many thanks in advance.

anhmiuhv, 2021-12-08:
I think you just try to ignore the warning.

you can use 

 GPIO.setwarnings(False)

to not print the warning

basfari, 2021-12-08:
Thank you for you quick answer. The warning says that the library is unlikely to work correctly. It is not true in my case, right (it will work correctly)?

anhmiuhv, 2021-12-08:
We don't test on 3rd party carrier board so I don't really know. But if the manufacturer said it is compatible, hopefully it should work ok.

closed 2022-01-30 by anhmiuhv