NVIDIA/jetson-gpio #65: I have installed the Jetson.GPIO on the 2GB jetson nano ,But the jupyter still remind me  No module named 'Jetson'.

Q455, 2022-01-29:
The output as follows:
[image]
My installing-process refer to this article:
https://pypi.org/project/Jetson.GPIO/

anhmiuhv, 2022-01-30:
Can you try to pip install inside the jupyter notebook
——————————
import sys
!{sys.executable} -m pip install Jetson.GPIO
——————————

Q455, 2022-01-30:
> Can you try to pip install inside the jupyter notebook —————————— import sys !{sys.executable} -m pip install Jetson.GPIO ——————————

Thank you for yuor useful advice ! But i still can't undersdand why such a convenient method is not mentioned on every artical about the installation of the Jetson.GPIO, instead illustrating such a complex way as i mentioned before,and why the complex way dose not work :https://pypi.org/project/Jetson.GPIO/

anhmiuhv, 2022-01-30:
It is most likely because your jupyter use a separate instance of python instead of the system’s default python. Note that you probably need to complete the “ Setting User Permissions” step in the https://pypi.org/project/Jetson.GPIO/

Q455, 2022-01-30:
> It is most likely because your jupyter use a separate instance of python instead of the system’s default python. Note that you probably need to complete the “ Setting User Permissions” step in the https://pypi.org/project/Jetson.GPIO/

Of course i follow the  “ Setting User Permissions” step ,But it still can't work .Maybe,your insight is right .My jupyter comes from the ip of the nano . All is default.

Q455, 2022-01-30:
A nice way to solve this problem : 
1. git clone https://github.com/NVIDIA/jetson-gipio
2. use the cod strightly in the Jupyter : 
                                                            pip install Jetson.GPIO

closed 2022-01-30 by Q455