NVIDIA/jetson-gpio #68: possible typo?

pjueon, 2022-04-13:
https://github.com/NVIDIA/jetson-gpio/blob/de68f76f0f423b8a4e5504d119446790b752a1a0/lib/python/Jetson/GPIO/gpio_pin_data.py#L430-L435

I'm not 100% sure, but shouldn't the line 435 in `gpio_pin_data.py` be `return s` instead of `return f` ?

anhmiuhv, 2022-04-18:
Thanks I will fix this

closed 2022-04-18 by anhmiuhv

pjueon, 2022-04-18:
@anhmiuhv 
Sorry if I'm wrong, but I feel like it should be `return s`, not `return ids` (ca3ccf09)
You iterated through `ids` and found `s`, which starts with `prefix`.  Then, shouldn't the return value be `s` instead of `ids`? 

In the line 426~429, the return value is the one which starts with `prefix`, not the entire file name list. 
So returning `ids` in the line 435 looks wrong to me. Or am I missing something?

anhmiuhv, 2022-04-18:
Yeah you are right