NVIDIA/jetson-gpio #74: GPIO pins not responding on Jetson ORIN

KTBM, 2022-07-13:
I've run both the test_all_pins.py and simple_out.py programs, the simple_out.py changed to all GPIO pins, and haven't seen a single instance in which a GPIO pin has moved from LOW to HIGH.
I am running on a jetson ORIN, new from the box with the newest jetpack. I've read on the Nvidia forums that JP5.0 has issues with the GPIO pins and was hoping someone could help and has a solution that doesn't require me to turn to the terminal or to sudo to change a GPIO pin output. 
L4T verison: 34.1.1
would appreciate any help

anhmiuhv, 2022-07-26:
If you can wait until the next release, it should be fixed.

anhmiuhv, 2022-07-26:
Otherwise, you can use the pinmux spreadsheet to set all the GPIO pins you need into output mode.

anhmiuhv, 2022-08-15:
New release is out!. It should fix this problem.

bnbhat, 2024-01-12:
I am experiencing the same issue where the GPIO pins are not responding at all. There are no errors during runtime, but none of the pins have toggled between HIGH and LOW.

I am also using the Orin Nano with a dev kit, running JetPack 6 and Jetson Linux 36.2, with kernel version 5.15.122-tegra.

Are there any specific steps I should follow?

I would highly appreciate any assistance you can provide. @KTBM @anhmiuhv

anhmiuhv, 2024-01-12:
Can you try to modify the pinmux of the output pin so that the output pin needs to be specified as output?

bnbhat, 2024-01-31:
Thank you so much for your reply!

I rolled back to JetPack 5.1.2, and it solved the issue for now.

But I will be transitioning to JetPack 6 again in the next month. I will definitely try modifying the Pinmux soon.

FedeMorenoOptima, 2024-03-03:
How you doing! I am facing the same problem when I use JetPack 6. I am not familiar with Pinmux... do you know when Jetpack 6 will be supported?
Thanks in advance
Federico.

anhmiuhv, 2024-03-03:
If you search pinmux on jetson downloads center
https://developer.nvidia.com/embedded/downloads#?search=pinmux

you can find the pinmux spreadsheet for the Jetson device that you have. Then for example you want to use this pin for Jetson AGX Orin

https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py#L102C1-L103C1

You can find pin  'PH.00' by searching in the spreadsheet. Then you can change the pin direction to 'Bidirectional'. After that, you can generate the pinmux dtb.

After you get the pinmux dtb you can flash it to your Jetson device. See this link fro more details https://elinux.org/Jetson/L4T_BSP_development_tips

anhmiuhv, 2024-09-16:
Close as duplicate to #114

closed 2024-09-16 by anhmiuhv