NVIDIA/jetson-gpio #79: TypeError: Can Only Concatenate str

Ecophagy, 2023-02-23:
Using the latest version 2.1.0 of Jetson.GPIO on a Jetson Nano with Python 3.8 (installed via pip), I encounter the fatal error "TypeError: Can only concatenate str" here: https://github.com/NVIDIA/jetson-gpio/blob/v2.1.0/lib/python/Jetson/GPIO/gpio_pin_data.py#L581

No special steps are required to reproduce - it occurs on `import Jetson.GPIO`.

As the error suggests, while `prefix` is a string, `pwm_chip_name` is not. I think this is because [this commit](https://github.com/NVIDIA/jetson-gpio/commit/1fd1e5f80c9b20e1595f311bacaa2bab3dade849) changed the definition of `JETSON_NANO_PIN_DEFS` to have each line contain 8 entries rather than 9 (by removing `{}` from each line). This means that the [iteration](https://github.com/NVIDIA/jetson-gpio/blob/v2.1.0/lib/python/Jetson/GPIO/gpio_pin_data.py#L577) to gather pwn_chip_names is now looking at the wrong field, which is an int instead of a string. A quick check looks like this will also apply to `JETSON_TX1_PIN_DEFS`

I would submit a patch, but I don't know what the correct fix is - most of the PIN_DEF arrays still have 9 entries, so possibly it's just re-adding the empty `{}` to keep them all the same shape? Or would it better to add a special case for 8-long PIN_DEFs? Possibly the commit author @anhmiuhv  could weight in?

It's unfortunate that this breaks the latest release - I think the only workaround is to pin at the previous version 2.0.21.

anhmiuhv, 2023-02-23:
Yes we are moving away from sysfs aproach so there should be a few bugs here and there. For Jetson Xavier and before you should be fine using version 2.0.21 for now.

Ecophagy, 2023-02-23:
Thanks for the swift response, I will stick with the stable version. 

Do I need to use the udev rules from that release, or is it safe to use the rules from 2.1.0?

anhmiuhv, 2023-02-23:
udev rules from that release

localJetson, 2023-03-12:
I was able to add str(pwm_chip_name) and it would throw out a different error in line 607 about indexing out of range in the tuple. I barely know any python so this is as far as I can understand

localJetson, 2023-03-13:
@Ecophagy 

So, I know this isn't the way to go but you can make the import work by correcting two lines of code within the "gpio_pin_data.py" file. The full path should be shown when you get the error so you just need to vim <path>

in line 581: change statement "d = prefix + pwm_chip_name" to "d = prefix + str(pwm_chip_name)"
in line 607: change statement "pwm_id=x[8]" to "pwm_id=x[7]"

Now, the lib worked for my use, i2c communication on pin 3 and 5 for Jetson Nano

I tried checking what the output of that function is if I changed the index it looked to see what information changed and I couldn't find any change. Again, limited python experience. Needed to make it work cause I didn't know how to import the lib to python2

anhmiuhv, 2023-04-17:
I think this is fixed in the latest version

closed 2023-04-17 by anhmiuhv