NVIDIA/jetson-gpio #80: Problem with miniconda

ferraridamiano, 2023-02-23:
Hello, I just followed the tutorial on the README to install the package and setup the permissions. I'm using miniconda. Unfortunately when I import the package I get the following error:
```
>>> import Jetson.GPIO as GPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/xavier/miniconda3/envs/yolo/lib/python3.8/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/home/xavier/miniconda3/envs/yolo/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 33, in <module>
    raise RuntimeError("The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that {} does not exist. Please check if that file is present.".format(_GPIOCHIP_ROOT))
RuntimeError: The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that /dev/gpiochip0 does not exist. Please check if that file is present
```
While if I import it with the native python environment all works good. Are there additional setup step to do for conda?

sleepat-11, 2023-03-01:
have you solved the problem? i use  archiconda ,i dont know how to install jetson gpio in it

ferraridamiano, 2023-03-01:
Unfortunately no