NVIDIA/jetson-gpio #82: Same permissions problem

sumitbounthiyal, 2023-03-29:
import Jetson.GPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/nvidia/.virtualenvs/cv/lib/python3.5/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/home/nvidia/.virtualenvs/cv/lib/python3.5/site-packages/Jetson/GPIO/gpio.py", line 33, in <module>
    raise RuntimeError("The current user does not have permissions set to "
RuntimeError: The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this

I am having the same problem with the code when I run sample code from File.
Please help me on this.

nghungthinh, 2023-03-29:
You can run command :
$ sudo python3 your_file.py 

in stead:
$ python3 your_file.py

sumitbounthiyal, 2023-03-29:
[image]
Same error is coming with python3 also

sumitbounthiyal, 2023-03-29:
My problem is similar like #2

mdnayeemsardar, 2023-03-30:
I am getting the same error. gotta do something... im gonna try rerun this package with its previous version. will let u know when things get working for me.. admin pls respond to this bug

anhmiuhv, 2023-04-17:
Did you install the udev rules before running the command

bhawkins-jpl, 2024-03-15:
I did add the user group and install the udev rules, but they didn't seem to work on my Xavier dev board. Changing the rule to
```bash
SUBSYSTEM=="gpio", KERNEL=="gpiochip*", ACTION=="add", \
    GROUP="gpio", MODE="0660"
```
seems to solve the problem. I'm not sure why the shell commands in the provided rule don't work.

anhmiuhv, 2024-03-16:
Ok I can look into that

closed 2025-02-06 by anhmiuhv