NVIDIA/jetson-gpio #86: Jetson Nano Gpio interrupt lagging

mohamed-keshk, 2023-06-16:
I am using the Jetson nano GPIO to read the output of an AB bi-phase rotational encoder encode, I wrote a simple python script to trigger a rising edge interrupt, the callback function prints the current state of the pin that triggered.
```
import time 
import Jetson.GPIO as GPIO 

# Define the GPIO pins for the rotary encoders 
PIN = 13

# Initialize GPIO GPIO.setmode(GPIO.BOARD) 
GPIO.setup(PIN, GPIO.IN) 
def callback(PIN): 
   print(GPIO.input(PIN))

# Add event detection for rising edges GPIO.add_event_detect(PIN, GPIO.RISING, callback=callback, bouncetime=1)

try: 
   while True: 
      pass 	 
except KeyboardInterrupt: 
   pass 

# Clean up GPIO 
GPIO.cleanup() 
```
After a lot of debugging and many many tries, I stopped using the actual encoder, I’m using a function generator set to output a 3.3v 500hz square wave on pin 13 on the Jetson nano. The output I expect should a stream of ones, but the actual output is random mix of ones and zeros.

The actual output from the terminal:
1
0
1
1
1
1
1
1
1
1
0
1
0
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
0
1
1
1
0
1
0
1
1
1
1

What am I missing or doing wrong.
This my system:
L4T 32.6.1 JetPack 4.6
Ubuntu 18.04.6 LTS
Kernel Version: 4.9.253-tegra

closed 2023-08-09 by anhmiuhv

23100048, 2024-03-01:
Having the same problem. Should be getting a constant stream of ones but get a mix of ones and zeros when the input signal is high. 
@mohamed-keshk, Can you please share any information on how to fix the issue?

anhmiuhv, 2024-03-01:
It is possible that the sampling rate is not high enough. In that case, you should write a c program to read sysfs or the gpio registers directly.

mohamed-keshk, 2024-03-01:
@23100048 I'm sorry to disappoint you but I couldn't find a satisfying answer to why that happened, nor have found a solution. I abandoned the project, I mean to get back to it some time later, but till now, I simply can't read the encoder reliably.
@anhmiuhv the sampling rate of the jetson nano gpio is a hell lot higher than the 500 hz signal I used to test it. If it can't handle such a slow signal, I guess we'd be better off using an uno.

anhmiuhv, 2024-03-01:
Yeah, it is likely that you will need to use real time kernel to reliably sample. Uno is probably easier

23100048, 2024-03-01:
Thankyou so much for your responses. Yes, I have shifted the process of reading the input signal to an arduino (mega 2560) and am communicating that via rosserial to the jetson nano. That has solved the problem.