NVIDIA/jetson-gpio #92: check HIGH or LOW state of the GPIO header

SJavad, 2023-08-09:
Hi
I have a Jetson TX1 board.
and I want to use one of the GPIO Header.
I have STM32 blue pill microcontroller 
and in code, I set one of the pins to get HIGH and LOW in 5 seconds
I want to connect the pin on the STM32 with a jumper wire to the jetson and in the jetson with Python, check if the GPIO pin that connects to the STM32 pin is HIGH or LOW.
how can I do that?

anhmiuhv, 2023-08-09:
There are examples you can follow

https://github.com/NVIDIA/jetson-gpio/blob/master/samples/button_interrupt.py
https://github.com/NVIDIA/jetson-gpio/blob/master/samples/simple_input.py
https://github.com/NVIDIA/jetson-gpio/blob/master/samples/simple_output.py

SJavad, 2023-08-10:
> There are examples you can follow
> 
> https://github.com/NVIDIA/jetson-gpio/blob/master/samples/button_interrupt.py https://github.com/NVIDIA/jetson-gpio/blob/master/samples/simple_input.py https://github.com/NVIDIA/jetson-gpio/blob/master/samples/simple_output.py

@anhmiuhv 
Hi, Thank you for your response 🌸🖐
is there no problem connecting the 3.3V jumper wire to jetson? 
I'm afraid to burn the board...
Does not want a resistor in the way of the jumper wire?

closed 2023-10-04 by anhmiuhv