NVIDIA/jetson-gpio #93: Pin32 of Orin Nano still does't work as PWM after reinstall new GPIO @2.1.3

alexzhangzt, 2023-08-31:
Hi Guys,
Pin32 of Orin Nano still does't work as PWM after reinstall new GPIO @2.1.3. But Pin15 and Pin33 can work well as PWM.
[image]
[image]
[image]
[image]
I use gpio-io.py to set Pin15,32,33 as PWM and reinstall new version 2.1.3 GPIO but Pin32 still doesn't work as PWM.
So I think that gpio_pin_data.py doesn't define JETSON_ORIN_NANO_PIN_DEFS. It is the reason.
Looking forward to resolving the BUG.

Thanks a ton

anhmiuhv, 2023-08-31:
what kernel version and what BSP version are you using?

anhmiuhv, 2023-09-01:
Can you get GPIO.VERSION also

ObamaIII, 2023-09-01:
Hi, Linh,
kernel version=Linux Kernel 5.10.104
BSP version = NVIDIA Jetson Linux 35.3.1
GPIO.VERSION = 2.1.3
JetPack 5.1.1
[image]

Thank you very much for your great help.
Best

ObamaIII, 2023-09-07:
> Can you get GPIO.VERSION also
Hi, Linh,
kernel version=Linux Kernel 5.10.104
BSP version = NVIDIA Jetson Linux 35.3.1
GPIO.VERSION = 2.1.3
JetPack 5.1.1

Thanks

anhmiuhv, 2023-09-07:
Also can you show me 

ls /sys/devices
ls/sys/devices/playform

ObamaIII, 2023-09-07:
> Also can you show me
> 
> ls /sys/devices ls/sys/devices/playform
GPIO.JETSON_INFO
[image]

ls /sys/devices
ls /sys/devices/playform
[image]

anhmiuhv, 2023-09-07:
oh I mean 

ls /sys/devices/platform

ObamaIII, 2023-09-07:
> oh I mean
> 
> ls /sys/devices/platform

[platform.txt](https://github.com/NVIDIA/jetson-gpio/files/12552460/platform.txt)

Thanks

ObamaIII, 2023-09-07:
> oh I mean
> 
> ls /sys/devices/platform

nvidia@tegra-ubuntu:~/mypro$ ls /sys/devices/platform
 10000000.iommu
 100000.chipid
 12000000.iommu
 12000000.smmu_suspend
 13800000.display
 13a00000.cbb-fabric
 13e40000.host1x
 140a0000.pcie
 14100000.pcie
 14160000.pcie
 141e0000.pcie
 168000000.ramoops_carveout
 17000000.ga10b
 2190000.watchdog
 2200000.gpio
 2430000.pinmux
 2600000.gpcdma
 2c00000.mc_sid
 2c10000.mc
 2c10000.mc-hwpm
 3100000.serial
 3130000.serial
 3140000.serial
 3160000.i2c
 3180000.i2c
 3190000.i2c
 31b0000.i2c
 31c0000.i2c
 31d0000.serial
 31e0000.i2c
 3210000.spi
 3230000.spi
 3280000.pwm
 32a0000.pwm
 32c0000.pwm
 3400000.sdhci
 3510000.hda
 3520000.xusb_padctl
 3550000.xudc
 3610000.xhci
 3810000.efuse
 39c0000.tachometer
 3ae0000.nvrng
 3c00000.tegra-hsp
 3c10000.combined-uart
 3d00000.tegra-hsp
 40000000.sysram
 8000000.iommu
 aconnect@2a41000
 arm-pmu
 b600000.sce-fabric
 b950000.tegra-hsp
 bc00000.rtcpu
 be00000.rce-fabric
 bpmp
 c000000.aon
 c150000.tegra-hsp
 c240000.i2c
 c250000.i2c
 c2a0000.rtc
 c2f0000.gpio
 c310000.mttcan
 c360000.pmc
 c600000.aon-fabric
 cam_i2cmux
 cbb
 cpuidle
 cpu-throttle-alert
 cv0-throttle-alert
 cv1-throttle-alert
 cv2-throttle-alert
 d230000.actmon
 d600000.bpmp-fabric
 d800000.dce
 de00000.dce-fabric
 dma_test
 dsu-pmu-0
 dsu-pmu-1
 dsu-pmu-2
 e000000.ccplex
 e100000.tegra_mce
 e860000.psc
 efi-framebuffer.0
 efivars.0
 efuse-burn
 f100000.tegra_soc_hwpm
 firmware:optee
'Fixed MDIO bus.0'
 fixed-regulators
 gpio-keys
 gpu.0
 gpu-throttle-alert
 host1x
 hot-surface-alert
 icc
 interrupt-controller
 mods-simple-bus
 nvguard_service_dramecc_diag_startup_test
 nvguard_service_pinmux_padring_diag_periodic_check
 nvguard_service_sample_1
 nvguard_service_sample_2
 nvguard_service_sample_3
 nvguard_service_sce_diag_hsmconfig_monitor
 nvguard_service_sce_diag_noc_bist
 nvguard_service_sce_diag_scelic_monitorerror
 nvguard_service_spi2_diag_e2e_check
 nvpmodel
 nvsciipc
 nvsciipc-kernel
 power
 psci
 psci-cpuidle
 pwm-fan
 reg-dummy
 regulatory.0
 scf-pmu
 serial8250
 smmu_test
 snd-soc-dummy
 soc0-throttle-alert
 soc1-throttle-alert
 soc2-throttle-alert
 soctherm-oc-event
 soft_watchdog
 sound
 spdif_dit
 tegra-cache
 tegra-camera-platform
 tegra-capture-vi
 tegra-carveouts
 tegra_fiq_debugger
 tegra-fuse
 timer
 uevent
 usb_cd

ObamaIII, 2023-09-07:
> 3280000.pwm
> 32a0000.pwm
> 32c0000.pwm
nvidia@tegra-ubuntu:~/mypro$ ls /sys/devices/platform：
[image]
In lib/python/Jetson/GPIO/gpio_pin_data.py:
  (85, 'PN.01', "tegra234-gpio", 15, 22, 'GPIO12', 'GP88_PWM1', '3280000.pwm', 0),
   (41, 'PG.06', "tegra234-gpio", 32, 12, 'GPIO07', 'GP113_PWM7', '32e0000.pwm', 0),
   (43, 'PH.00', "tegra234-gpio", 33, 13, 'GPIO13', 'GP115', '32c0000.pwm', 0),
[image]
[image]
Looks like .pwm is same in platform and gpio_pin_data.py, but JETSON_ORIN_NX_PIN_DEFS is NOT compats_jetson_orins_nano = (
seems like  JETSON_ORIN_NANO_PIN_DEFS is NOT Defined in /gpio_pin_data.py.

ObamaIII, 2023-09-07:
> (41, 'PG.06', "tegra234-gpio", 32, 12, 'GPIO07', 'GP113_PWM7', '32e0000.pwm', 0),

need to change '32e0000.pwm' into '32a0000.pwm'  IN (41, 'PG.06', "tegra234-gpio", 32, 12, 'GPIO07', 'GP113_PWM7', '32e0000.pwm', 0), right?

ObamaIII, 2023-09-07:
> > (41, 'PG.06', "tegra234-gpio", 32, 12, 'GPIO07', 'GP113_PWM7', '32e0000.pwm', 0),
> 
> need to change '32e0000.pwm' into '32a0000.pwm' IN (41, 'PG.06', "tegra234-gpio", 32, 12, 'GPIO07', 'GP113_PWM7', '32e0000.pwm', 0), right?

[image]

anhmiuhv, 2023-09-07:
32a0000.pwm is the fan. I read the release note again and found out that pin32 as PWM is only supported on BSP 35.4.1

anhmiuhv, 2023-09-07:
So probably need to update your BSP to 35.4.1 or avoid using pin 32 as pwm

ObamaIII, 2023-09-09:
> 32e0000.p

closed 2023-09-09 by ObamaIII

ObamaIII, 2023-09-09:
> So probably need to update your BSP to 35.4.1 or avoid using pin 32 as pwm

Hi,
Thank you very much for your great help.
I will try to update my BSP to 35.4.1. Do I copy only 32e0000.pwm from BSP 35.4.1 to my  /sys/devices/platform? because there is no 32e0000.pwm in the directory.
Best