NVIDIA/jetson-gpio #95: Python NameError: name 'open' is no defined. error.

yijh0611, 2023-10-30:
**Please don't submit questions**
Please use this form to report specific bugs in Jetson.GPIO. All general questions or support requests should be directed at NVIDIA's forums; see:
https://developer.nvidia.com/embedded/community/support-resources

Please delete this section from your bug report.

**Describe the bug**
A clear and concise description of what the bug is, including
- Which version of Jetson.GPIO you are using.
- Which version of L4T (not JetPack) you are using.
- Which Jetson board you are using.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect  button/wire/LED/... XXX to the board
2. Configure pin ... to input/output/...
3. Execute API ...

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.
--- --- ---
HW info
GPIO version 2.1.1 and 2.1.4
Jetpack 5.1.2
Linux 5.10.120-tegra
L4T 35.4.1

GPIO.JETSON_INFO
{'P1_REVISION': 1, 'RAM': '32768M, 65536M', 'REVISION': 'Unknown', 'TYPE': 'JETSON_ORIN_NX', 'MANUFACTURER': 'NVIDIA', 'PROCESSOR': 'A78AE'}

--- --- ---
Hello,

I encountered a runtime warning and an exception when attempting to initialize PWM on GPIO using Python. Here's the code snippet that triggers the error:

pwm = GPIO.PWM(32, 50) # 50hz

Upon execution, the following error is produced:

/usr/local/lib/python3.8/dist-packages/Jetson.GPIO-2.1.4-py3.8.egg/Jetson/GPIO/gpio_event.py:182: RuntimeWarning: Event not found
Exception ignored in: <function PWM.__del__ at 0xffffb6e91e50>
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/Jetson.GPIO-2.1.4-py3.8.egg/Jetson/GPIO/gpio.py", line 610, in __del__
  File "/usr/local/lib/python3.8/dist-packages/Jetson.GPIO-2.1.4-py3.8.egg/Jetson/GPIO/gpio.py", line 203, in _unexport_pwm
NameError: name 'open' is not defined

The issue persists across Python versions 3.8.10, 3.8.18, and 3.9.18. Could you please advise on how to resolve this?

Thank you for your assistance.

anhmiuhv, 2023-10-30:
Did you follow the example here to use PWM
https://github.com/NVIDIA/jetson-gpio/blob/master/samples/simple_pwm.py

Can you post your full code

yijh0611, 2023-10-30:
This is the full code.

``` Python
# import RPi.GPIO as GPIO
import Jetson.GPIO as GPIO
import time

def callback(data):
    pass

# init Servo
SERVO_PIN = 33 # 32

GPIO.setwarnings(False)
GPIO.setmode(GPIO.BOARD)
GPIO.setup(SERVO_PIN, GPIO.OUT, initial=GPIO.HIGH)

pwm = GPIO.PWM(SERVO_PIN, 50) # 50hz
```
--- --- ---
I used Jetson.GPIO becuase I encountered the following error.
```
RuntimeError: This module can only be run on a Raspberry Pi!
```

The code runs fine when the callback function is commented out. Additionally, the simple_pwm.py code works without issues. I will adjust my code based on the example you provided. Thank you.

anhmiuhv, 2023-10-30:
Yes I believe you will need to run GPIO.cleanup at the end of any Jetson.GPIO program

yijh0611, 2023-10-31:
> Yes I believe you will need to run GPIO.cleanup at the end of any Jetson.GPIO program

Thank you. GPIO.cleanup was the problem.

closed 2023-10-31 by yijh0611